open-mmlab/OpenPCDet #1160: The provided pre-trained PointPillar test metrics on the kitti test set differed significantly from those in the paper

MLbeginner2, 2022-10-28:
I submitted the official OpenPCDet results of pointpillar_7728.pth on the KITTI test set to KITTI, but the metrics are far below the values in the paper, is it because OpenPCDet's PointPillar only used 3712 samples during training? Do I need to change any parameters if I want to train with all the training samples?
The following figure shows the results of PointPillar on the KITTI test set：
[image]

Townjj, 2022-11-04:
It's not official code of Pointpillar.

closed 2022-12-20 by github-actions[bot]

bqcs, 2024-01-12:
> I submitted the official OpenPCDet results of pointpillar_7728.pth on the KITTI test set to KITTI, but the metrics are far below the values in the paper, is it because OpenPCDet's PointPillar only [...]

Have you identified the cause of the problem?