open-mmlab/OpenPCDet #1161: Test my PVRCNN in kitti,there is just bev mAP

Rody1911641, 2022-11-01:
This is my result:
Car AP@0.70, 0.70, 0.70:
bbox AP:0.0000, 0.0000, 0.0000
bev  AP:77.9522, 65.4295, 64.4617
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:0.00, 0.00, 0.00
Car AP_R40@0.70, 0.70, 0.70:
bbox AP:0.0000, 0.0000, 0.0000
bev  AP:79.5713, 66.7626, 64.5092
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:0.00, 0.00, 0.00
Car AP@0.70, 0.50, 0.50:
bbox AP:0.0000, 0.0000, 0.0000
bev  AP:89.4668, 86.4132, 86.2657
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:0.00, 0.00, 0.00
Car AP_R40@0.70, 0.50, 0.50:
bbox AP:0.0000, 0.0000, 0.0000
bev  AP:93.9974, 88.5902, 88.6612
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:0.00, 0.00, 0.00

This is my config:
CLASS_NAMES: ['Car']

DATA_CONFIG:
    _BASE_CONFIG_: cfgs/dataset_configs/kitti_car_DA.yaml
    CLASS_NAMES: ['Car']
    FOV_POINTS_ONLY: True
    SHIFT_COOR: [0.0, 0.0, 1.8]
    

MODEL:
    NAME: PVRCNNPlusPlus

    VFE:
        NAME: MeanVFE

    BACKBONE_3D:
        NAME: VoxelBackBone8x

    MAP_TO_BEV:
        NAME: HeightCompression
        NUM_BEV_FEATURES: 256

    BACKBONE_2D:
        NAME: BaseBEVBackbone

        LAYER_NUMS: [5, 5]
        LAYER_STRIDES: [1, 2]
        NUM_FILTERS: [128, 256]
        UPSAMPLE_STRIDES: [1, 2]
        NUM_UPSAMPLE_FILTERS: [256, 256]

    DENSE_HEAD:
        NAME: AnchorHeadSingle
        CLASS_AGNOSTIC: False

        USE_DIRECTION_CLASSIFIER: True
        DIR_OFFSET: 0.78539
        DIR_LIMIT_OFFSET: 0.0
        NUM_DIR_BINS: 2

        ANCHOR_GENERATOR_CONFIG: [
            {
                'class_name': 'car',
                'anchor_sizes': [[4.7, 2.1, 1.7]],
                'anchor_rotations': [0, 1.57],
                'anchor_bottom_heights': [0],
                'align_center': False,
                'feature_map_stride': 8,
                'matched_threshold': 0.55,
                'unmatched_threshold': 0.4
            }
        ]

        TARGET_ASSIGNER_CONFIG:
            NAME: AxisAlignedTargetAssigner
            POS_FRACTION: -1.0
            SAMPLE_SIZE: 512
            NORM_BY_NUM_EXAMPLES: False
            MATCH_HEIGHT: False
            BOX_CODER: ResidualCoder

        LOSS_CONFIG:
            LOSS_WEIGHTS: {
                'cls_weight': 1.0,
                'loc_weight': 2.0,
                'dir_weight': 0.2,
                'code_weights': [1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0]
            }

    PFE:
        NAME: VoxelSetAbstraction
        POINT_SOURCE: raw_points
        NUM_KEYPOINTS: 2048
        NUM_OUTPUT_FEATURES: 90
        SAMPLE_METHOD: SPC
        SPC_SAMPLING:
            NUM_SECTORS: 6
            SAMPLE_RADIUS_WITH_ROI: 1.6

        FEATURES_SOURCE: ['bev', 'x_conv3', 'x_conv4', 'raw_points']
        SA_LAYER:
            raw_points:
                NAME: VectorPoolAggregationModuleMSG
                NUM_GROUPS: 2
                LOCAL_AGGREGATION_TYPE: local_interpolation
                NUM_REDUCED_CHANNELS: 1
                NUM_CHANNELS_OF_LOCAL_AGGREGATION: 32
                MSG_POST_MLPS: [ 32 ]
                FILTER_NEIGHBOR_WITH_ROI: True
                RADIUS_OF_NEIGHBOR_WITH_ROI: 2.4

                GROUP_CFG_0:
                    NUM_LOCAL_VOXEL: [ 2, 2, 2 ]
                    MAX_NEIGHBOR_DISTANCE: 0.2
                    NEIGHBOR_NSAMPLE: -1
                    POST_MLPS: [ 32, 32 ]
                GROUP_CFG_1:
                    NUM_LOCAL_VOXEL: [ 3, 3, 3 ]
                    MAX_NEIGHBOR_DISTANCE: 0.4
                    NEIGHBOR_NSAMPLE: -1
                    POST_MLPS: [ 32, 32 ]

            x_conv3:
                DOWNSAMPLE_FACTOR: 4
                INPUT_CHANNELS: 64

                NAME: VectorPoolAggregationModuleMSG
                NUM_GROUPS: 2
                LOCAL_AGGREGATION_TYPE: local_interpolation
                NUM_REDUCED_CHANNELS: 32
                NUM_CHANNELS_OF_LOCAL_AGGREGATION: 32
                MSG_POST_MLPS: [128]
                FILTER_NEIGHBOR_WITH_ROI: True
                RADIUS_OF_NEIGHBOR_WITH_ROI: 4.0

                GROUP_CFG_0:
                    NUM_LOCAL_VOXEL: [3, 3, 3]
                    MAX_NEIGHBOR_DISTANCE: 1.2
                    NEIGHBOR_NSAMPLE: -1
                    POST_MLPS: [64, 64]
                GROUP_CFG_1:
                    NUM_LOCAL_VOXEL: [ 3, 3, 3 ]
                    MAX_NEIGHBOR_DISTANCE: 2.4
                    NEIGHBOR_NSAMPLE: -1
                    POST_MLPS: [ 64, 64 ]

            x_conv4:
                DOWNSAMPLE_FACTOR: 8
                INPUT_CHANNELS: 64

                NAME: VectorPoolAggregationModuleMSG
                NUM_GROUPS: 2
                LOCAL_AGGREGATION_TYPE: local_interpolation
                NUM_REDUCED_CHANNELS: 32
                NUM_CHANNELS_OF_LOCAL_AGGREGATION: 32
                MSG_POST_MLPS: [ 128 ]
                FILTER_NEIGHBOR_WITH_ROI: True
                RADIUS_OF_NEIGHBOR_WITH_ROI: 6.4

                GROUP_CFG_0:
                    NUM_LOCAL_VOXEL: [ 3, 3, 3 ]
                    MAX_NEIGHBOR_DISTANCE: 2.4
                    NEIGHBOR_NSAMPLE: -1
                    POST_MLPS: [ 64, 64 ]
                GROUP_CFG_1:
                    NUM_LOCAL_VOXEL: [ 3, 3, 3 ]
                    MAX_NEIGHBOR_DISTANCE: 4.8
                    NEIGHBOR_NSAMPLE: -1
                    POST_MLPS: [ 64, 64 ]

    POINT_HEAD:
        NAME: PointHeadSimple
        CLS_FC: [256, 256]
        CLASS_AGNOSTIC: True
        USE_POINT_FEATURES_BEFORE_FUSION: True
        TARGET_CONFIG:
            GT_EXTRA_WIDTH: [0.2, 0.2, 0.2]
        LOSS_CONFIG:
            LOSS_REG: smooth-l1
            LOSS_WEIGHTS: {
                'point_cls_weight': 1.0,
            }

    ROI_HEAD:
        NAME: PVRCNNHead
        CLASS_AGNOSTIC: True

        SHARED_FC: [256, 256]
        CLS_FC: [256, 256]
        REG_FC: [256, 256]
        DP_RATIO: 0.3

        NMS_CONFIG:
            TRAIN:
                NMS_TYPE: nms_gpu
                MULTI_CLASSES_NMS: False
                NMS_PRE_MAXSIZE: 9000
                NMS_POST_MAXSIZE: 512
                NMS_THRESH: 0.8
            TEST:
                NMS_TYPE: nms_gpu
                MULTI_CLASSES_NMS: False
                NMS_PRE_MAXSIZE: 1024
                NMS_POST_MAXSIZE: 100
                NMS_THRESH: 0.7
                SCORE_THRESH: 0.1

        ROI_GRID_POOL:
            GRID_SIZE: 6

            NAME: VectorPoolAggregationModuleMSG
            NUM_GROUPS: 2
            LOCAL_AGGREGATION_TYPE: voxel_random_choice
            NUM_REDUCED_CHANNELS: 30
            NUM_CHANNELS_OF_LOCAL_AGGREGATION: 32
            MSG_POST_MLPS: [ 128 ]

            GROUP_CFG_0:
                NUM_LOCAL_VOXEL: [ 3, 3, 3 ]
                MAX_NEIGHBOR_DISTANCE: 0.8
                NEIGHBOR_NSAMPLE: 32
                POST_MLPS: [ 64, 64 ]
            GROUP_CFG_1:
                NUM_LOCAL_VOXEL: [ 3, 3, 3 ]
                MAX_NEIGHBOR_DISTANCE: 1.6
                NEIGHBOR_NSAMPLE: 32
                POST_MLPS: [ 64, 64 ]

        TARGET_CONFIG:
            BOX_CODER: ResidualCoder
            ROI_PER_IMAGE: 128
            FG_RATIO: 0.5

            SAMPLE_ROI_BY_EACH_CLASS: True
            CLS_SCORE_TYPE: roi_iou

            CLS_FG_THRESH: 0.75
            CLS_BG_THRESH: 0.25
            CLS_BG_THRESH_LO: 0.1
            HARD_BG_RATIO: 0.8

            REG_FG_THRESH: 0.55

        LOSS_CONFIG:
            CLS_LOSS: BinaryCrossEntropy
            REG_LOSS: smooth-l1
            CORNER_LOSS_REGULARIZATION: True
            LOSS_WEIGHTS: {
                'rcnn_cls_weight': 1.0,
                'rcnn_reg_weight': 1.0,
                'rcnn_corner_weight': 1.0,
                'code_weights': [1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0]
            }

    POST_PROCESSING:
        RECALL_THRESH_LIST: [0.3, 0.5, 0.7]
        SCORE_THRESH: 0.1
        OUTPUT_RAW_SCORE: False

        EVAL_METRIC: kitti

        NMS_CONFIG:
            MULTI_CLASSES_NMS: False
            NMS_TYPE: nms_gpu
            NMS_THRESH: 0.1
            NMS_PRE_MAXSIZE: 4096
            NMS_POST_MAXSIZE: 500

OPTIMIZATION:
    OPTIMIZER: adam_onecycle
    LR: 0.01
    WEIGHT_DECAY: 0.001
    MOMENTUM: 0.9

    MOMS: [0.95, 0.85]
    PCT_START: 0.4
    DIV_FACTOR: 10
    DECAY_STEP_LIST: [35, 45]
    LR_DECAY: 0.1
    LR_CLIP: 0.0000001

    LR_WARMUP: False
    WARMUP_EPOCH: 1

    GRAD_NORM_CLIP: 10

Rody1911641, 2022-11-01:
2022-11-01 03:23:22,191   INFO  **********************Start logging**********************
2022-11-01 03:23:22,192   INFO  CUDA_VISIBLE_DEVICES=ALL
2022-11-01 03:23:22,192   INFO  cfg_file         cfgs/kitti_models/DAfaster.yaml
2022-11-01 03:23:22,192   INFO  batch_size       24
2022-11-01 03:23:22,192   INFO  workers          4
2022-11-01 03:23:22,192   INFO  extra_tag        pvrcnn++_intensity_ROS
2022-11-01 03:23:22,192   INFO  ckpt             /opt/data/private/3D-Detection/OpenPCDet/output/nuscenes_models/DAfaster/pvrcnn++_intensity_ROS/ckpt/checkpoint_epoch_30.pth
2022-11-01 03:23:22,192   INFO  pretrained_model None
2022-11-01 03:23:22,192   INFO  launcher         none
2022-11-01 03:23:22,193   INFO  tcp_port         18888
2022-11-01 03:23:22,193   INFO  local_rank       0
2022-11-01 03:23:22,193   INFO  set_cfgs         None
2022-11-01 03:23:22,193   INFO  max_waiting_mins 30
2022-11-01 03:23:22,193   INFO  start_epoch      0
2022-11-01 03:23:22,193   INFO  eval_tag         default
2022-11-01 03:23:22,193   INFO  eval_all         False
2022-11-01 03:23:22,193   INFO  ckpt_dir         None
2022-11-01 03:23:22,193   INFO  save_to_file     False
2022-11-01 03:23:22,193   INFO  infer_time       False
2022-11-01 03:23:22,194   INFO  cfg.ROOT_DIR: /opt/data/private/3D-Detection/OpenPCDet
2022-11-01 03:23:22,194   INFO  cfg.LOCAL_RANK: 0
2022-11-01 03:23:22,194   INFO  cfg.CLASS_NAMES: ['Car']
2022-11-01 03:23:22,194   INFO  
cfg.DATA_CONFIG = edict()
2022-11-01 03:23:22,194   INFO  cfg.DATA_CONFIG.DATASET: KittiDataset
2022-11-01 03:23:22,194   INFO  cfg.DATA_CONFIG.DATA_PATH: ../data/kitti
2022-11-01 03:23:22,194   INFO  cfg.DATA_CONFIG.POINT_CLOUD_RANGE: [-75.2, -75.2, -2, 75.2, 75.2, 4]
2022-11-01 03:23:22,194   INFO  
cfg.DATA_CONFIG.DATA_SPLIT = edict()
2022-11-01 03:23:22,194   INFO  cfg.DATA_CONFIG.DATA_SPLIT.train: train
2022-11-01 03:23:22,194   INFO  cfg.DATA_CONFIG.DATA_SPLIT.test: val
2022-11-01 03:23:22,195   INFO  
cfg.DATA_CONFIG.INFO_PATH = edict()
2022-11-01 03:23:22,195   INFO  cfg.DATA_CONFIG.INFO_PATH.train: ['kitti_infos_train.pkl']
2022-11-01 03:23:22,195   INFO  cfg.DATA_CONFIG.INFO_PATH.test: ['kitti_infos_val.pkl']
2022-11-01 03:23:22,195   INFO  
cfg.DATA_CONFIG.DATA_AUGMENTOR = edict()
2022-11-01 03:23:22,195   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.DISABLE_AUG_LIST: ['random_object_scaling', 'random_object_rotation']
2022-11-01 03:23:22,195   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.AUG_CONFIG_LIST: [{'NAME': 'random_object_scaling', 'SCALE_UNIFORM_NOISE': [0.9, 1.1]}, {'NAME': 'random_object_rotation', 'ROT_PROB': 1.0, 'ROT_UNIFORM_NOISE': [-0.78539816, 0.78539816]}, {'NAME': 'random_world_flip', 'ALONG_AXIS_LIST': ['x', 'y']}, {'NAME': 'random_world_rotation', 'WORLD_ROT_ANGLE': [-0.78539816, 0.78539816]}, {'NAME': 'random_world_scaling', 'WORLD_SCALE_RANGE': [0.95, 1.05]}]
2022-11-01 03:23:22,195   INFO  
cfg.DATA_CONFIG.POINT_FEATURE_ENCODING = edict()
2022-11-01 03:23:22,195   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.encoding_type: absolute_coordinates_encoding
2022-11-01 03:23:22,195   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.used_feature_list: ['x', 'y', 'z', 'intensity']
2022-11-01 03:23:22,195   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.src_feature_list: ['x', 'y', 'z', 'intensity']
2022-11-01 03:23:22,196   INFO  cfg.DATA_CONFIG.DATA_PROCESSOR: [{'NAME': 'mask_points_and_boxes_outside_range', 'REMOVE_OUTSIDE_BOXES': True}, {'NAME': 'shuffle_points', 'SHUFFLE_ENABLED': {'train': True, 'test': False}}, {'NAME': 'transform_points_to_voxels', 'VOXEL_SIZE': [0.1, 0.1, 0.15], 'MAX_POINTS_PER_VOXEL': 5, 'MAX_NUMBER_OF_VOXELS': {'train': 80000, 'test': 90000}}]
2022-11-01 03:23:22,196   INFO  
cfg.DATA_CONFIG.TEST = edict()
2022-11-01 03:23:22,196   INFO  
cfg.DATA_CONFIG.TEST.BOX_FILTER = edict()
2022-11-01 03:23:22,196   INFO  cfg.DATA_CONFIG.TEST.BOX_FILTER.USE_IMAGE_AREA_FILTER: True
2022-11-01 03:23:22,196   INFO  cfg.DATA_CONFIG.TEST.BOX_FILTER.FOV_FILTER: True
2022-11-01 03:23:22,196   INFO  cfg.DATA_CONFIG.TEST.BOX_FILTER.LIMIT_RANGE: [-75.2, -75.2, -2, 75.2, 75.2, 4]
2022-11-01 03:23:22,196   INFO  cfg.DATA_CONFIG._BASE_CONFIG_: cfgs/dataset_configs/kitti_car_DA.yaml
2022-11-01 03:23:22,196   INFO  cfg.DATA_CONFIG.CLASS_NAMES: ['Car']
2022-11-01 03:23:22,196   INFO  cfg.DATA_CONFIG.FOV_POINTS_ONLY: True
2022-11-01 03:23:22,196   INFO  cfg.DATA_CONFIG.SHIFT_COOR: [0.0, 0.0, 1.8]
2022-11-01 03:23:22,196   INFO  cfg.DATA_CONFIG.MAX_SWEEPS: 1
2022-11-01 03:23:22,196   INFO  cfg.DATA_CONFIG.PRED_VELOCITY: False
2022-11-01 03:23:22,197   INFO  cfg.DATA_CONFIG.BALANCED_RESAMPLING: False
2022-11-01 03:23:22,197   INFO  
cfg.MODEL = edict()
2022-11-01 03:23:22,197   INFO  cfg.MODEL.NAME: PVRCNNPlusPlus
2022-11-01 03:23:22,197   INFO  
cfg.MODEL.VFE = edict()
2022-11-01 03:23:22,197   INFO  cfg.MODEL.VFE.NAME: MeanVFE
2022-11-01 03:23:22,197   INFO  
cfg.MODEL.BACKBONE_3D = edict()
2022-11-01 03:23:22,197   INFO  cfg.MODEL.BACKBONE_3D.NAME: VoxelBackBone8x
2022-11-01 03:23:22,197   INFO  
cfg.MODEL.MAP_TO_BEV = edict()
[... 271 lines cut ...]
@numba.jit(nopython=True, parallel=True)
def d3_box_overlap_kernel(boxes, qboxes, rinc, criterion=-1):
^

  state.func_ir.loc))
2022-11-01 03:31:34,528   INFO  Car AP@0.70, 0.70, 0.70:
bbox AP:0.0000, 0.0000, 0.0000
bev  AP:77.9522, 65.4295, 64.4617
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:0.00, 0.00, 0.00
Car AP_R40@0.70, 0.70, 0.70:
bbox AP:0.0000, 0.0000, 0.0000
bev  AP:79.5713, 66.7626, 64.5092
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:0.00, 0.00, 0.00
Car AP@0.70, 0.50, 0.50:
bbox AP:0.0000, 0.0000, 0.0000
bev  AP:89.4668, 86.4132, 86.2657
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:0.00, 0.00, 0.00
Car AP_R40@0.70, 0.50, 0.50:
bbox AP:0.0000, 0.0000, 0.0000
bev  AP:93.9974, 88.5902, 88.6612
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:0.00, 0.00, 0.00

closed 2022-11-02 by Rody1911641

liyn69sysu, 2023-12-04:
hello, can you teach me how to solve this problem

Rody-NKCS, 2023-12-05:
你好，我已经不做这个方向了，凭借记忆，我记得是因为，我在训练时对点云z坐标加上了一定的偏移，而测试时忘记加上同样的偏移，所以bev视角下测试结果是对的，但3DAP就不对了。