open-mmlab/OpenPCDet #1168: how to run demo with docker?

WuYanXingege, 2022-11-05:
i find docker hasn't tools/demo.py

jihanyang, 2022-11-06:
I haven't try to run openpcdet on docker. Hope others can help you.

Townjj, 2022-11-11:
I have run demo.py in docker successfully. demo.py don't exist in docker may be the file sync proble between CONTAINER and LOCALHOST. Check your script when build your container:

-v /LocalDir:ContainerDir

MooooNSC, 2022-11-16:
you need GUI setting

Try 

nvidia-docker run -it --gpus 'all,"capabilities=compute,utility,graphics"' --name &{name} -e DISPLAY=unix$DISPLAY --volume="/etc/group:/etc/group:ro" --volume="/etc/passwd:/etc/passwd:ro" --volume="/etc/shadow:/etc/shadow:ro" --volume="/etc/sudoers.d:/etc/sudoers.d:ro" --volume="/tmp/.X11-unix:/tmp/.X11-unix:rw" &{IMAGE_ID}

vikasnr, 2022-12-05:
> 
When I try to run demo.py it can't import pcdet

`from pcdet.config import cfg, cfg_from_yaml_file
ModuleNotFoundError: No module named 'pcdet'
`

In the [`installation guide`](https://github.com/open-mmlab/OpenPCDet/blob/master/docs/INSTALL.md) as per the note
pcdet needs to installed.
"NOTE: Please re-install` pcdet v0.5` by running `python setup.py develop` even if you have already installed previous version."

Did you install pcdet? How did you do it through Dockerfile?

closed 2023-01-19 by github-actions[bot]

ShioHZY, 2023-05-09:
> > 
> 
> @Townjj @MooooNSC @WuYanXingege
> 
> When I try to run demo.py it can't import pcdet
> 
> `from pcdet.config import cfg, cfg_from_yaml_file ModuleNotFoundError: No module named 'pcdet' `
> 
> In the [`installation guide`](https://github.com/open-mmlab/OpenPCDet/blob/master/docs/INSTALL.md) as per the note pcdet needs to installed.
> 
> "NOTE: Please re-install `pcdet v0.5` by running `python setup.py develop` even if you have already installed previous version."
> 
> Did you install pcdet? How did you do it through Dockerfile?
> 
> Any idea how to run docker in non-interactive mode (installation of dependencies like pcdet has to be through Dockerfile)?

@vikasnr 
hi, i met the same problem. i pull the docker image directly, and i can't import pcdet. should i install it by myself? have you solved it? any reply is appreciated.