open-mmlab/OpenPCDet #1171: Reproduce PillarNet on Waymo val dataset

yanghtr, 2022-11-06:
I directly used `OpenPCDet/tools/cfgs/waymo_models/pillarnet.yaml` to train PillarNet on the Waymo dataset (batch size 32). The results are worse than the reported numbers in the paper. Any ideas to reproduce the results?

My results:
```
2022-11-05 21:16:28,977   INFO  **********************Start evaluation waymo_models/pillarnet(default)**********************
2022-11-05 21:16:28,982   INFO  Loading Waymo dataset
2022-11-05 21:16:55,771   INFO  Total skipped info 0
2022-11-05 21:16:55,771   INFO  Total samples for Waymo dataset: 39987
2022-11-05 21:16:55,793   INFO  ==> Loading parameters from checkpoint /***/OpenPCDet/output/waymo_models/pillarnet/default/ckpt/checkpoint_epoch_30.pth to CPU
2022-11-05 21:16:55,959   INFO  ==> Checkpoint trained from version: pcdet+0.6.0+f221374
2022-11-05 21:16:55,996   INFO  ==> Done (loaded 307/307)
2022-11-05 21:16:56,008   INFO  *************** EPOCH 30 EVALUATION *****************
2022-11-05 21:31:48,042   INFO  *************** Performance of EPOCH 30 *****************
2022-11-05 21:31:48,043   INFO  Generate label finished(sec_per_example: 0.0223 second).
2022-11-05 21:31:48,043   INFO  recall_roi_0.3: 0.000000
2022-11-05 21:31:48,043   INFO  recall_rcnn_0.3: 0.864593
2022-11-05 21:31:48,043   INFO  recall_roi_0.5: 0.000000
2022-11-05 21:31:48,043   INFO  recall_rcnn_0.5: 0.795496
2022-11-05 21:31:48,043   INFO  recall_roi_0.7: 0.000000
2022-11-05 21:31:48,043   INFO  recall_rcnn_0.7: 0.517223
2022-11-05 21:31:48,070   INFO  Average predicted number of objects(39987 samples): 216.503
2022-11-06 06:28:52,973   INFO
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/AP: 0.7222
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/APH: 0.7166
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/APL: 0.7222
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2/AP: 0.6421
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2/APH: 0.6369
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2/APL: 0.6421
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1/AP: 0.7187
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1/APH: 0.6303
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1/APL: 0.7187
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2/AP: 0.6410
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2/APH: 0.5608
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2/APL: 0.6410
OBJECT_TYPE_TYPE_SIGN_LEVEL_1/AP: 0.0000
OBJECT_TYPE_TYPE_SIGN_LEVEL_1/APH: 0.0000
OBJECT_TYPE_TYPE_SIGN_LEVEL_1/APL: 0.0000
OBJECT_TYPE_TYPE_SIGN_LEVEL_2/AP: 0.0000
OBJECT_TYPE_TYPE_SIGN_LEVEL_2/APH: 0.0000
OBJECT_TYPE_TYPE_SIGN_LEVEL_2/APL: 0.0000
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1/AP: 0.6724
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1/APH: 0.6591
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1/APL: 0.6724
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2/AP: 0.6474
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2/APH: 0.6346
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2/APL: 0.6474

```

closed 2023-01-02 by github-actions[bot]

zhangkangkai, 2023-02-23:
I have the same problem as you. Have you solved it? @yanghtr

shawnding, 2023-04-15:
From https://github.com/open-mmlab/OpenPCDet#waymo-open-dataset-baselines:

> By default, all models are trained with a single frame of 20% data (~32k frames) of all the training samples on 8 GTX 1080Ti GPUs, and the results of each cell here are mAP/mAPH calculated by the official Waymo evaluation metrics on the whole validation set (version 1.2).

Did you set the `SAMPLED_INTERVAL` to 1 when reproduce the result? Thanks!