open-mmlab/OpenPCDet #1174: custom data training error

hariura, 2022-11-08:
After run following code, i faced the error 

python -m pcdet.datasets.custom.custom_dataset create_custom_infos tools/cfgs/dataset_configs/custom_dataset.yaml

train sample_idx: 012402
train sample_idx: 004523
Custom info train file is save to /home/jh/Desktop/3d/OpenPCDet/data/custom/custom_infos_train.pkl
Traceback (most recent call last):
  File "/home/jh/anaconda3/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/home/jh/anaconda3/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/jh/Desktop/3d/OpenPCDet/pcdet/datasets/custom/custom_dataset.py", line 278, in <module>
    create_custom_infos(
  File "/home/jh/Desktop/3d/OpenPCDet/pcdet/datasets/custom/custom_dataset.py", line 255, in create_custom_infos
    custom_infos_val = dataset.get_infos(
  File "/home/jh/Desktop/3d/OpenPCDet/pcdet/datasets/custom/custom_dataset.py", line 164, in get_infos
    infos = executor.map(process_single_scene, sample_id_list)
  File "/home/jh/anaconda3/lib/python3.9/concurrent/futures/_base.py", line 598, in map
    fs = [self.submit(fn, *args) for args in zip(*iterables)]
TypeError: 'NoneType' object is not iterable

TypeError: 'NoneType' object is not iterable

-------------------------------------
i think there is problem in custom_dataset.py in my folder 
but i don't know what is reason  

        dataset_cfg = EasyDict(yaml.safe_load(open(sys.argv[2])))
        ROOT_DIR = (Path(__file__).resolve().parent / '../../../').resolve()
        create_custom_infos(
            dataset_cfg=dataset_cfg,
            class_names=['Car', 'Pedestrian', 'Truck'],
            data_path=ROOT_DIR / 'data' / 'custom',
            save_path=ROOT_DIR / 'data' / 'custom',
        )

hariura, 2022-11-08:
thank you for answering my question. 
I have .pcd  format file so i convert pcd file to npy 
I have tried to run demo.py and it work well. 
so i don't think pcd data is wrong. but im not sure about label 

I have psr data label
I'm tring to find correct imformation,but I can't find information about heading angle
so i use rotation value about z which is about 3 and -3 ( x,y data is zero)

it is my sample raw label data 

{'obj_id': '211',
 'obj_type': 'Truck',
 'psr': {'position': {'x': -15.970201300989684,
   'y': 3.289973588268538,
   'z': -0.15352165699005116},
  'rotation': {'x': 0, 'y': 0, 'z': -3.115412714809878},
  'scale': {'x': 9.502174582745123,
   'y': 2.580846390739077,
   'z': 3.090679168701172}}}
   

I convert json to txt

it is my 000067.txt file in labels

0.202729 -4.064130 -0.984596 4.146975 1.838396 1.225871 -3.123521 Car
15.453259 4.045643 -1.581176 5.050665 1.859315 1.830762 3.115388 Truck
27.183142 -0.253994 -0.758320 2.249685 2.410332 3.285114 -3.063064 Truck
22.231623 -4.442561 -2.038527 1.803613 1.906385 0.972601 -3.063060 Car
7.830323 -4.009229 -1.630289 4.420130 1.823082 1.188379 -3.115397 Car
7.855523 -0.162319 -1.518514 3.712430 1.869286 1.256537 -3.063053 Car
24.509953 3.443007 -2.378261 0.468776 1.658684 0.892412 3.101957 Car

I'm sorry for my little understanding, and thanks for all opinion

closed 2023-01-14 by github-actions[bot]

saurtone, 2023-02-14:
Are you successful now?

saurtone, 2023-02-14:
邮件我已收到。辛苦了。

swaggywilliam, 2023-12-09:
Hello, I have met the same error with you but I used the kitti_dataset.py
```
Traceback (most recent call last):
  File "/home/anaconda3/envs/casa/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/home/anaconda3/envs/casa/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/project/CasA/pcdet/datasets/kitti/kitti_dataset.py", line 442, in <module>
    create_kitti_infos(
  File "/home/project/CasA/pcdet/datasets/kitti/kitti_dataset.py", line 406, in create_kitti_infos
    kitti_infos_train = dataset.get_infos(num_workers=workers, has_label=True, count_inside_pts=True)
  File "/home/project/CasA/pcdet/datasets/kitti/kitti_dataset.py", line 190, in get_infos
    infos = executor.map(process_single_scene, sample_id_list)
  File "/home/anaconda3/envs/casa/lib/python3.9/concurrent/futures/_base.py", line 598, in map
    fs = [self.submit(fn, *args) for args in zip(*iterables)]
TypeError: 'NoneType' object is not iterable
```
Have you solved it?

saurtone, 2023-12-09:
邮件我已收到。辛苦了。

swaggywilliam, 2023-12-09:
> Hello, I have met the same error with you but I used the kitti_dataset.py
>
> ```
> Traceback (most recent call last):
>   File "/home/anaconda3/envs/casa/lib/python3.9/runpy.py", line 197, in [...]

I have solved it, the 'data_path' in kitti_dataset.py should be update to your data_path

Harry710887048, 2024-03-10:
> I have solved it, the 'data_path' in kitti_dataset.py should be update to your data_path

What does it mean, can you be more specific ?

saurtone, 2024-03-10:
邮件我已收到。辛苦了。

esther-ET, 2024-03-10:
> What does it mean, can you be more specific ?

Hi! If you had solved it, could you tell me how？ Thank you!

Harry710887048, 2024-03-12:
> Hi! If you had solved it, could you tell me how？ Thank you!

I don't think it's necessarily a path issue because I've checked and there doesn't seem to be any problem with the paths. I suspect it might be due to the version of OpenPCDet. When running the MODEST code(https://github.com/YurongYou/MODEST), I encountered the error using pcdet0.3 as required, but when I ran python -m pcdet.datasets.kitti.kitti_dataset create_kitti_infos from MODEST's downstream/OpenPCDet, there were no errors.

Harry710887048, 2024-03-12:
> Hi! If you had solved it, could you tell me how？ Thank you!

I don't think it's necessarily a path issue because I've checked and there doesn't seem to be any problem with the paths. I suspect it might be due to the version of OpenPCDet. When running the MODEST code(https://github.com/YurongYou/MODEST), I encountered the error using pcdet0.3 as required when I ran python -m pcdet.datasets.kitti.kitti_dataset create_kitti_infos，then I used pcdet in MODEST(downstream/OpenPCDet), there were no errors.

esther-ET, 2024-03-12:
Thank you very much. I think so and I
have something wrong with the dataset document , so I have the problem with no iterable. By the way , how do you arrange your data document ? Thanks for your reply!

     

 Traceback (most recent call last):   File "/home/anaconda3/envs/casa/lib/python3.9/runpy.py", line 197, in _run_module_as_main     return _run_code(code, main_globals, None,   File "/home/anaconda3/envs/casa/lib/python3.9/runpy.py", line 87, in _run_code     exec(code, run_globals)   File "/home/project/CasA/pcdet/datasets/kitti/kitti_dataset.py", line 442, in <module&gt;     create_kitti_infos(   File "/home/project/CasA/pcdet/datasets/kitti/kitti_dataset.py", line 406, in create_kitti_infos     kitti_infos_train = dataset.get_infos(num_workers=workers, has_label=True, count_inside_pts=True)   File "/home/project/CasA/pcdet/datasets/kitti/kitti_dataset.py", line 190, in get_infos     infos = executor.map(process_single_scene, sample_id_list)   File "/home/anaconda3/envs/casa/lib/python3.9/concurrent/futures/_base.py", line 598, in map     fs = [self.submit(fn, *args) for args in zip(*iterables)] TypeError: 'NoneType' object is not iterable

swaggywilliam, 2024-03-12:
> I don't think it's necessarily a path issue because I've checked and there doesn't seem to be any problem with the paths. I suspect it might be due to the version of OpenPCDet. When running the [...]

I just modified the data path and the issue was solved, I don't know why it don't work with you.

Harry710887048, 2024-03-12:
> Thank you very much. I think so and I have something wrong with the dataset document , so I have the problem with no iterable. By the way , how do you arrange your data document ? Thanks for your reply!
> […](#)

[image]
[image]
I 'm running on the nuscenes dataset.

esther-ET, 2024-03-12:
Thank you！！！ 

  

  
image.png (view on web)
 image.png (view on web)

liuyuansheng624, 2024-07-15:
I can train and test on openpcdet using custom point cloud datasets
weichat13731082126

saurtone, 2024-07-15:
邮件我已收到。辛苦了。

Cipher-zzz, 2024-09-07:
This can be solved by adding the 'ImageSets' folder and test/train/val.txt followed by the other datasets folder.

saurtone, 2024-09-07:
邮件我已收到。辛苦了。

esther-ET, 2024-09-07:
OK, thanks, I have solved it.

lxzzzzz, 2024-09-14:
I don't know what your issue, but my issue is no test.txt in ImageSets folder. Hope useful to u~

esther-ET, 2024-09-14:
I have solved it. Thanks！