open-mmlab/OpenPCDet #1176: RuntimeError: Error compiling objects for extension

Clichong, 2022-11-09:
When I try "python setup.py develop", it was wrong as following:

running develop
/home/lab/anaconda3/envs/openpcdet/lib/python3.7/site-packages/setuptools/command/easy_install.py:147: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  EasyInstallDeprecationWarning,
/home/lab/anaconda3/envs/openpcdet/lib/python3.7/site-packages/setuptools/command/install.py:37: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  setuptools.SetuptoolsDeprecationWarning,
running egg_info
writing pcdet.egg-info/PKG-INFO
writing dependency_links to pcdet.egg-info/dependency_links.txt
writing requirements to pcdet.egg-info/requires.txt
writing top-level names to pcdet.egg-info/top_level.txt
reading manifest file 'pcdet.egg-info/SOURCES.txt'
adding license file 'LICENSE'
writing manifest file 'pcdet.egg-info/SOURCES.txt'
running build_ext
building 'pcdet.ops.iou3d_nms.iou3d_nms_cuda' extension
creating /home/lab/LLC/PointCloud/OpenPCDet/build
creating /home/lab/LLC/PointCloud/OpenPCDet/build/temp.linux-x86_64-cpython-37
creating /home/lab/LLC/PointCloud/OpenPCDet/build/temp.linux-x86_64-cpython-37/pcdet
creating /home/lab/LLC/PointCloud/OpenPCDet/build/temp.linux-x86_64-cpython-37/pcdet/ops
creating /home/lab/LLC/PointCloud/OpenPCDet/build/temp.linux-x86_64-cpython-37/pcdet/ops/iou3d_nms
creating /home/lab/LLC/PointCloud/OpenPCDet/build/temp.linux-x86_64-cpython-37/pcdet/ops/iou3d_nms/src
Emitting ninja build file /home/lab/LLC/PointCloud/OpenPCDet/build/temp.linux-x86_64-cpython-37/build.ninja...
Compiling objects...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
[1/4] /usr/local/cuda/bin/nvcc --generate-dependencies-with-compile --dependency-output /home/lab/LLC/PointCloud/OpenPCDet/build/temp.linux-x86_64-cpython-37/pcdet/ops/iou3d_nms/src/iou3d_nms_kernel.o.d -I/home/lab/anaconda3/envs/openpcdet/lib/python3.7/site-packages/torch/include -I/home/lab/anaconda3/envs/openpcdet/lib/python3.7/site-packages/torch/include/torch/csrc/api/include -I/home/lab/anaconda3/envs/openpcdet/lib/python3.7/site-packages/torch/include/TH -I/home/lab/anaconda3/envs/openpcdet/lib/python3.7/site-packages/torch/include/THC -I/usr/local/cuda/include -I/home/lab/anaconda3/envs/openpcdet/include/python3.7m -c -c /home/lab/LLC/PointCloud/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_nms_kernel.cu -o /home/lab/LLC/PointCloud/OpenPCDet/build/temp.linux-x86_64-cpython-37/pcdet/ops/iou3d_nms/src/iou3d_nms_kernel.o -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_BFLOAT16_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr --compiler-options ''"'"'-fPIC'"'"'' -DTORCH_API_INCLUDE_EXTENSION_H '-DPYBIND11_COMPILER_TYPE="_gcc"' '-DPYBIND11_STDLIB="_libstdcpp"' '-DPYBIND11_BUILD_ABI="_cxxabi1011"' -DTORCH_EXTENSION_NAME=iou3d_nms_cuda -D_GLIBCXX_USE_CXX11_ABI=0 -gencode=arch=compute_86,code=compute_86 -gencode=arch=compute_86,code=sm_86 -std=c++14
FAILED: /home/lab/LLC/PointCloud/OpenPCDet/build/temp.linux-x86_64-cpython-37/pcdet/ops/iou3d_nms/src/iou3d_nms_kernel.o
/usr/local/cuda/bin/nvcc --generate-dependencies-with-compile --dependency-output /home/lab/LLC/PointCloud/OpenPCDet/build/temp.linux-x86_64-cpython-37/pcdet/ops/iou3d_nms/src/iou3d_nms_kernel.o.d -I/home/lab/anaconda3/envs/openpcdet/lib/python3.7/site-packages/torch/include -I/home/lab/anaconda3/envs/openpcdet/lib/python3.7/site-packages/torch/include/torch/csrc/api/include -I/home/lab/anaconda3/envs/openpcdet/lib/python3.7/site-packages/torch/include/TH -I/home/lab/anaconda3/envs/openpcdet/lib/python3.7/site-packages/torch/include/THC -I/usr/local/cuda/include -I/home/lab/anaconda3/envs/openpcdet/include/python3.7m -c -c /home/lab/LLC/PointCloud/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_nms_kernel.cu -o /home/lab/LLC/PointCloud/OpenPCDet/build/temp.linux-x86_64-cpython-37/pcdet/ops/iou3d_nms/src/iou3d_nms_kernel.o -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_BFLOAT16_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr --compiler-options ''"'"'-fPIC'"'"'' -DTORCH_API_INCLUDE_EXTENSION_H '-DPYBIND11_COMPILER_TYPE="_gcc"' '-DPYBIND11_STDLIB="_libstdcpp"' '-DPYBIND11_BUILD_ABI="_cxxabi1011"' -DTORCH_EXTENSION_NAME=iou3d_nms_cuda -D_GLIBCXX_USE_CXX11_ABI=0 -gencode=arch=compute_86,code=compute_86 -gencode=arch=compute_86,code=sm_86 -std=c++14
/bin/sh: 1: /usr/local/cuda/bin/nvcc: not found
[2/4] c++ -MMD -MF /home/lab/LLC/PointCloud/OpenPCDet/build/temp.linux-x86_64-cpython-37/pcdet/ops/iou3d_nms/src/iou3d_cpu.o.d -pthread -B /home/lab/anaconda3/envs/openpcdet/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -I/home/lab/anaconda3/envs/openpcdet/lib/python3.7/site-packages/torch/include -I/home/lab/anaconda3/envs/openpcdet/lib/python3.7/site-packages/torch/include/torch/csrc/api/include -I/home/lab/anaconda3/envs/openpcdet/lib/python3.7/site-packages/torch/include/TH -I/home/lab/anaconda3/envs/openpcdet/lib/python3.7/site-packages/torch/include/THC -I/usr/local/cuda/include -I/home/lab/anaconda3/envs/openpcdet/include/python3.7m -c -c /home/lab/LLC/PointCloud/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_cpu.cpp -o /home/lab/LLC/PointCloud/OpenPCDet/build/temp.linux-x86_64-cpython-37/pcdet/ops/iou3d_nms/src/iou3d_cpu.o -DTORCH_API_INCLUDE_EXTENSION_H '-DPYBIND11_COMPILER_TYPE="_gcc"' '-DPYBIND11_STDLIB="_libstdcpp"' '-DPYBIND11_BUILD_ABI="_cxxabi1011"' -DTORCH_EXTENSION_NAME=iou3d_nms_cuda -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++14
cc1plus: warning: command-line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
In file included from /home/lab/anaconda3/envs/openpcdet/lib/python3.7/site-packages/torch/include/ATen/Parallel.h:140,
                 from /home/lab/anaconda3/envs/openpcdet/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/utils.h:3,
                 from /home/lab/anaconda3/envs/openpcdet/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/nn/cloneable.h:5,
                 from /home/lab/anaconda3/envs/openpcdet/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/nn.h:3,
                 from /home/lab/anaconda3/envs/openpcdet/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/all.h:13,
                 from /home/lab/anaconda3/envs/openpcdet/lib/python3.7/site-packages/torch/include/torch/extension.h:4,
                 from /home/lab/LLC/PointCloud/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_cpu.cpp:10:
/home/lab/anaconda3/envs/openpcdet/lib/python3.7/site-packages/torch/include/ATen/ParallelOpenMP.h:83: warning: ignoring ‘#pragma omp parallel’ [-Wunknown-pragmas]
   83 | #pragma omp parallel for if ((end - begin) >= grain_size)
      |
/home/lab/LLC/PointCloud/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_cpu.cpp: In function ‘int boxes_iou_bev_cpu(at::Tensor, at::Tensor, at::Tensor)’:
/home/lab/LLC/PointCloud/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_cpu.cpp:242:54: warning: ‘T* at::Tensor::data() const [with T = float]’ is deprecated: Tensor.data<T>() is deprecated. Please use Tensor.data_ptr<T>() instead. [-Wdeprecated-declarations]
  242 |     const float *boxes_a = boxes_a_tensor.data<float>();
      |                            ~~~~~~~~~~~~~~~~~~~~~~~~~~^~
In file included from /home/lab/anaconda3/envs/openpcdet/lib/python3.7/site-packages/torch/include/ATen/Tensor.h:3,
                 from /home/lab/anaconda3/envs/openpcdet/lib/python3.7/site-packages/torch/include/ATen/Context.h:4,
                 from /home/lab/anaconda3/envs/openpcdet/lib/python3.7/site-packages/torch/include/ATen/ATen.h:9,
                 from /home/lab/anaconda3/envs/openpcdet/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/types.h:3,
                 from /home/lab/anaconda3/envs/openpcdet/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/serialize/input-archive.h:6,
                 from /home/lab/anaconda3/envs/openpcdet/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/serialize/archive.h:3,
                 from /home/lab/anaconda3/envs/openpcdet/lib/python3.7/site-packages/torch/include/torch/csrc/api/include/torch/serialize/tensor.h:3,
                 from /home/lab/LLC/PointCloud/OpenPCDet/pcdet/ops/iou3d_nms/src/iou3d_cpu.cpp:9:
[... cut ...]
  File "/home/lab/anaconda3/envs/openpcdet/lib/python3.7/site-packages/torch/utils/cpp_extension.py", line 538, in unix_wrap_ninja_compile
    with_cuda=with_cuda)
  File "/home/lab/anaconda3/envs/openpcdet/lib/python3.7/site-packages/torch/utils/cpp_extension.py", line 1359, in _write_ninja_file_and_compile_objects
    error_prefix='Error compiling objects for extension')
  File "/home/lab/anaconda3/envs/openpcdet/lib/python3.7/site-packages/torch/utils/cpp_extension.py", line 1683, in _run_ninja_build
    raise RuntimeError(message) from e
RuntimeError: Error compiling objects for extension

cls1206, 2022-11-11:
the same to you

cls1206, 2022-11-11:
Hi, I have the same issue. How to solve this problem?

Clichong, 2022-11-11:
> 

I haven't solved this yet. I had tried several cuda or pytorch versions, but they all failed.

Clichong, 2022-11-11:
> Hi, I have the same issue. How to solve this problem?

I haven't solved this yet. I had tried several cuda or pytorch versions, but they all failed.

ShaohuaDong2021, 2022-11-19:
Same question, have you solved this problem?

Clichong, 2022-11-20:
> Same question, have you solved this problem?

not yet

ShaohuaDong2021, 2022-11-20:
> 
I have solved this problem by creating a new environment:
conda create -n [new env]
conda install pytorch==1.9.0 torchvision==0.10.0 torchaudio==0.9.0 cudatoolkit=10.2 -c pytorch
pip install spconv
python setup.py develop

Clichong, 2022-11-20:
> 

I ran your configuration, but it still fails. My server is 3090, and I have tried multiple cuda and pytorch versions, all of which failed.

Clichong, 2022-11-20:
> > 
> 
> I have solved this problem by creating a new environment: conda create -n [new env] conda install pytorch==1.9.0 torchvision==0.10.0 torchaudio==0.9.0 cudatoolkit=10.2 -c pytorch pip install spconv python setup.py develop

I ran your configuration, but it still fails. My server is 3090, and I have tried multiple cuda and pytorch versions, all of which failed.

Clichong, 2022-11-20:
I feel that OpenPCDet is not compatible with 3090 server

closed 2023-01-04 by github-actions[bot]