open-mmlab/OpenPCDet #1178: Trouble with custom dataset evaluation, 3d and bev AP is 0

Encycloquandia, 2022-11-10:
First I would like to thank you for this project, I had a query regarding training using our own dataset. 
My annotations are corresponding to the axis of the point cloud, and I use cenerpoint as my model. But when evaluation, the results of 3d and bev AP are ZERO, while bbox and aos are not zero, and i don't know why. The screenshot is below.

[image]

And i use dmo.py to visualize the result, the blue boxs are the gt_boxes and the green boxes are the pred_boxes, I wonder why the green boxes are neat and well above the ground.

[image]

Is there anyone who could guide me as to how to do it,
Thanks for your  help!!!

Encycloquandia, 2022-11-10:
有没有哪位大佬可以教教！使用官方提供的custom代码训练自己数据时候，在评估的时候发现3d和bev指标都是0，而2d的指标是有数值的，使用demo.py可视化的结果发现预测的框是整齐的一排，而且是远高于地面的，是因为我的标签设置有问题吗？但是模型参数不对呢？

SweetBittErz, 2022-11-23:
Hello, I want to realize the evaluation of custom data set during training, but the program reports an error. I want to guide how to modify evaluation () function in custom_ dataset.py allows it to evaluate custom datasets. My dataset is labeled according to the kitti format, but I do not have calib, so you can understand that, except for xyzwhl class and angle, all other parameters are 0

SweetBittErz, 2022-12-24:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。

S-kewen, 2022-12-30:
Hello, I have a similar problem, we try to label our dataset (KITTI-format) in Carla simulator for evaluation, but we found that the AP results are all 0.00

S-kewen, 2023-03-01:
> > First I would like to thank you for this project, I had a query regarding training using our own dataset. My annotations are corresponding to the axis of the point cloud, and I use cenerpoint as my model. But when evaluation, the results of 3d and bev AP are ZERO, while bbox and aos are not zero, and i don't know why. The screenshot is below.
> > [image]
> > And i use dmo.py to visualize the result, the blue boxs are the gt_boxes and the green boxes are the pred_boxes, I wonder why the green boxes are neat and well above the ground.
> > [image]
> > Is there anyone who could guide me as to how to do it, Thanks for your help!!!
> 
> 你成功了吗，我也想训练自己的点云数据。

没有，训练多次失败之后放弃了

ghb0224, 2023-03-14:
> > > 首先，我要感谢您的这个项目，我有一个关于使用我们自己的数据集进行训练的查询。我的注释对应于点云的轴，我使用cenerpoint作为我的模型。但是在评估时，3d和bev AP的结果为零，而bbox和aos的结果不为零，我不知道为什么。截图如下。 [image]我用 dmo.py 来可视化结果，蓝色的盒子是gt_boxes的，绿色的盒子是pred_boxes的，我想知道为什么绿色的盒子很整齐，远远高于地面。 [image]有没有人可以指导我如何做，谢谢你的帮助!!!
> > 
> > 
> > 你成功了吗，我也想训练自己的点云数据。
> 
> 没有，训练多次失败之后放弃了
你好，我想查看我的评价指标，该如何操作？

ghb0224, 2023-03-14:
> First I would like to thank you for this project, I had a query regarding training using our own dataset. My annotations are corresponding to the axis of the point cloud, and I use cenerpoint as my [...]

你好，我想查看我的评价指标，该如何操作？目前已经训练、验证完成

closed 2023-04-28 by github-actions[bot]