open-mmlab/OpenPCDet #1181: Error when I use demo.py to visualize result. How can I add images as input?

ChunZheng2022, 2022-11-14:
Hello, error occurs when I use demo.py to visualize my multimodal  
result. How can I read images in demo.py

2022-11-04 15:19:40,312 INFO ==> Loading parameters from checkpoint ../output/kitti_models/fspconv80.pth to CPU
2022-11-04 15:19:40,362 INFO ==> Checkpoint trained from version: pcdet+0.6.0+0000000
2022-11-04 15:19:40,411 INFO ==> Done (loaded 653/653)
2022-11-04 15:19:40,586 INFO Visualized sample index: 1
Traceback (most recent call last):
File "demo.py", line 112, in
main()
File "demo.py", line 98, in main
pred_dicts, _ = model.forward(data_dict)
File "/home/lcz/code/platform/openpcdet/pcdet/models/detectors/voxel_rcnn.py", line 11, in forward
batch_dict = cur_module(batch_dict)
File "/home/lcz/anaconda3/envs/cu113/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
return forward_call(*input, **kwargs)
File "/home/lcz/code/platform/openpcdet/pcdet/models/backbones_3d/spconv_backbone_focal.py", line 236, in forward
x_image = self.semseg(batch_dict['images'])['layer1_feat2d']
KeyError: 'images'

meiyihTan, 2022-11-22:
I also facing the exactly same issue here, is there any idea to solve this?

ChunZheng2022, 2022-11-22:
> I also facing the exactly same issue here, is there any idea to solve this?

Hi,
test.py is availabe for multimodal input. You can use test.py to get predict result and then visualize it yourself.
If using demo.py is required, I think you can try rewrite DemoDataset in demo.py.

Mi97ch, 2022-11-29:
Do you mean that I can use test.py to detect objects on custom data?

ChunZheng2022, 2022-12-04:
> Do you mean that I can use test.py to detect objects on custom data?

Yes, but you need to organize your data in kitti format

closed 2023-01-18 by github-actions[bot]