open-mmlab/OpenPCDet #1187: About train

VsionQing, 2022-11-16:
/home/hello/.conda/envs/mmd/bin/python /data/Open/OpenPCDet-master/tools/train.py --cfg_file=cfgs/kitti_models/pv-rcnn.yaml--batch_size=4
Traceback (most recent call last):
  File "/data/Open/OpenPCDet-master/tools/train.py", line 7, in <module>
    from test import repeat_eval_ckpt
  File "/data/Open/OpenPCDet-master/tools/test.py", line 14, in <module>
    from eval_utils import eval_utils
  File "/data/Open/OpenPCDet-master/tools/eval_utils/eval_utils.py", line 8, in <module>
    from pcdet.models import load_data_to_gpu
  File "/data/Open/OpenPCDet-master/tools/../pcdet/models/__init__.py", line 6, in <module>
    from .detectors import build_detector
  File "/data/Open/OpenPCDet-master/tools/../pcdet/models/detectors/__init__.py", line 1, in <module>
    from .detector3d_template import Detector3DTemplate
  File "/data/Open/OpenPCDet-master/tools/../pcdet/models/detectors/detector3d_template.py", line 6, in <module>
    from ...ops.iou3d_nms import iou3d_nms_utils
  File "/data/Open/OpenPCDet-master/tools/../pcdet/ops/iou3d_nms/iou3d_nms_utils.py", line 9, in <module>
    from . import iou3d_nms_cuda
ImportError: /data/Open/OpenPCDet-master/tools/../pcdet/ops/iou3d_nms/iou3d_nms_cuda.cpython-38-x86_64-linux-gnu.so: undefined symbol: _ZNK3c1010TensorImpl36is_contiguous_nondefault_policy_implENS_12MemoryFormatE

VsionQing, 2022-11-17:
I configured the virtual environment according to the requirements file, but I still reported this error. Can someone help me solve the following problems

Townjj, 2022-11-26:
Did you successfully run >python setup.py develop ?

VsionQing, 2022-11-26:
收到谢谢

VsionQing, 2022-11-26:
> 

Yes, it has run successfully
I installed the latest version of pcdet to solve this problem

VsionQing, 2022-12-27:
收到谢谢

VsionQing, 2023-01-27:
收到谢谢

VsionQing, 2023-02-27:
收到谢谢

closed 2023-04-15 by github-actions[bot]