open-mmlab/OpenPCDet #1194: Why can't test after deleting the training set

Frank-DA, 2022-11-21:
When I delete the training set, the running test will report an error.

Original Traceback (most recent call last):
  File "/home/xidian/anaconda3/envs/pro/lib/python3.6/site-packages/torch/utils/data/_utils/worker.py", line 287, in _worker_loop
    data = fetcher.fetch(index)
  File "/home/xidian/anaconda3/envs/pro/lib/python3.6/site-packages/torch/utils/data/_utils/fetch.py", line 49, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/xidian/anaconda3/envs/pro/lib/python3.6/site-packages/torch/utils/data/_utils/fetch.py", line 49, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "../pcdet/datasets/kitti/kitti_dataset.py", line 380, in __getitem__
    calib = self.get_calib(sample_idx)
  File "../pcdet/datasets/kitti/kitti_dataset.py", line 110, in get_calib
    assert calib_file.exists()
AssertionError

Can you help me to analyze the reason?

Frank-DA, 2022-11-22:
I found that the code used training sets for testing.

Townjj, 2022-11-26:
Try to change the pkl file path in dataset config.

jihanyang, 2022-11-27:
You actually delete the validation set, not the training set.

closed 2023-01-11 by github-actions[bot]