open-mmlab/OpenPCDet #1197: How to implement evaluation for customdataset

SweetBittErz, 2022-11-23:
Hello, I want to realize the evaluation of custom data set during training, but the program reports an error. I want to guide how to modify evaluation () function in custom_ dataset.py allows it to evaluate custom datasets. My dataset is labeled according to the kitti format, but I do not have calib, so you can understand that, except for xyzwhl class and angle, all other parameters are 0.
I copy the official evaltion() function which in custom_dataset.py to my custom_ dataset.py, and then run it directly. The following error occurs:

2022-11-23 14:57:08,407   INFO  **********************Start evaluation custom_models/pointrcnn(default)**********************
2022-11-23 14:57:08,408   INFO  Loading Custom dataset.
2022-11-23 14:57:08,409   INFO  Total samples for CUSTOM dataset: 74
2022-11-23 14:57:08,411   INFO  ==> Loading parameters from checkpoint /home/Downloads/OpenPCDet/OpenPCDet/output/custom_models/pointrcnn/default/ckpt/checkpoint_epoch_500.pth to GPU
2022-11-23 14:57:08,443   INFO  ==> Checkpoint trained from version: pcdet+0.6.0+f221374
2022-11-23 14:57:08,455   INFO  ==> Done (loaded 309/309)
2022-11-23 14:57:08,456   INFO  *************** EPOCH 500 EVALUATION *****************
eval: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 37/37 [00:11<00:00,  3.25it/s, recall_0.3=(87, 88) / 90]
2022-11-23 14:57:19,855   INFO  *************** Performance of EPOCH 500 *****************
2022-11-23 14:57:19,855   INFO  Generate label finished(sec_per_example: 0.1540 second).
2022-11-23 14:57:19,855   INFO  recall_roi_0.3: 0.966667
2022-11-23 14:57:19,855   INFO  recall_rcnn_0.3: 0.977778
2022-11-23 14:57:19,855   INFO  recall_roi_0.5: 0.888889
2022-11-23 14:57:19,855   INFO  recall_rcnn_0.5: 0.944444
2022-11-23 14:57:19,855   INFO  recall_roi_0.7: 0.377778
2022-11-23 14:57:19,855   INFO  recall_rcnn_0.7: 0.544444
2022-11-23 14:57:19,855   INFO  Average predicted number of objects(37 samples): 1.054
Traceback (most recent call last):
  File "train.py", line 227, in <module>
    main()
  File "train.py", line 220, in main
    dist_test=dist_train
  File "/home/Downloads/OpenPCDet/OpenPCDet/tools/test.py", line 125, in repeat_eval_ckpt
    result_dir=cur_result_dir
  File "/home/Downloads/OpenPCDet/OpenPCDet/tools/eval_utils/eval_utils.py", line 128, in eval_one_epoch
    output_path=final_output_dir
  File "../pcdet/datasets/custom/custom_dataset.py", line 80, in evaluation
    ap_result_str, ap_dict = kitti_eval(eval_det_annos, eval_gt_annos, self.map_class_to_kitti)
AttributeError: 'CustomDataset' object has no attribute 'map_class_to_kitti'

Although my dataset label is in kitti format, there is no calib file and image, so the corresponding values are all 0. How can I modify these functions to evaluate my dataset？

jihanyang, 2022-11-24:
Please read the tutorial carefully.
https://github.com/open-mmlab/OpenPCDet/blob/master/docs/CUSTOM_DATASET_TUTORIAL.md

closed 2022-11-24 by jihanyang

SweetBittErz, 2022-11-24:
1.Of course I read this file，but it still reported this error when I added this in the yaml file. 
2. So I changed the method and copied the evaluation () function in kitti, but a new error occurred :len(gt_annos) different from len(dt_annos),I debug and found that there are two more dictionaries in dt_annos than in gt_annos（score and alpha）,How can I correct this problem? Whether to add these two dictionaries in gt_annos or delete them in dt_annos？

SweetBittErz, 2022-11-24:
I have now completed the create_infos、training and demo.py and can run successfully, but I cannot modify the evaluation to see the loss curve and evaluation index

jihanyang, 2022-11-24:
Have you provide `map_class_to_kitti` in the config?

SweetBittErz, 2022-11-25:
Yes，i have provide map_class_to_kitti in the config，but it reported a new error. I think it may be because of my custom_ dataset.py is modified by kitti_dataset.py(Not modify directly in custom_dataset.py which in the repo)。Now the problem I want to solve now is how to calculate the loss curve AP/mAP、IOU and other indicators to complete my graduation design. I hope I can get your help。
I have two ideas. Both of them have made mistakes. I want to ask you which is feasible. The implementation log is as follows:
1. modify kitti_dataset.py to custom_dataset.py
   This idea is implemented according to this blog（https://blog.csdn.net/JulyLi2019/article/details/126351276）. All functions other than evaluation have been implemented。when i debugged, i found that the reason why i could not evaluate was that the custom_val_annos.pkl is missing,so i add it in custom_dataset which also modified by kitti_dataset.py, but after generate this file,the program generate the following error:

so i debug to watch the difference between gt_annos and dt_annos:

Later i also tried Complete the dict in gt_annos，but It may be that my data does not have other data in the kitti data format（）. I think it is not feasible to directly run kitti eval，because when I create data, I use the OpenPCDet standard format, but save it in the kitti format. That is, in my Label, the fields with more kitti formats than the standard format are all 0.
So I want to ask you how to solve this problem? I have an idea. Can you tell me which functions can be used to generate loss curves and AP/MAP, IOU or other indicators, and can I directly use them instead of evaluation () in train.py?

2. use the custom_dataset.py in repo directly
but i don't know how to modify it  concrete,Could you tell me more about the specific work

Thank you from the bottom of my heart. I am very anxious for the evaluation results. I hope you can help me. I will be very grateful。

jihanyang, 2022-11-26:
If you just want to record the loss, it is already recorded in the log and tensorboard. As for the mAP, we don't calculate the performance metric during training as it will cause a lot of time, but you can check the code in eval_utils.py if you want to add such feature.

SweetBittErz, 2022-11-26:
Thank you very much, i have directly tried the complete modification method in repo. It can also work, but the following problems occurred during the evaluation. I hope you can help me。
1. When running train.py, the program will be stuck as soon as the evaluation part is reached。（But the first time I ran it, there were results，i will show you after. So I think this modification method which in repo is simple and feasible. What I learned in csdn before is complex.Thank you for your team work!!! This is exciting！！！）

2. Now i will show you the Screenshot of the only run result：

I have a question about this,because it only have the part result. I guess it may be because I have only one class,so i modify map_to_kitti to only {'ship':'Car'}. I want to ask you whether this is the reason, and whether the evaluation result of ship is correct after my modification. If I don't want the other two classes, how do I modify them?

I want to ask you whether this is the reason, and whether the evaluation result of ship is correct after my modification. If I don't want the other two classes, how do I modify them?

I think if these two problems can be solved, it should be a success. Thank you for your great team！！Looking forward to hearing from you.

SweetBittErz, 2022-11-26:
Finally, I have another question. My training has been completed, but when I want to view Loss in tensorboard, the following error appears. How can I solve it?

And  this is the train.py's log，It's stuck in the evaluation. I don't know why, but when an updated pth is not generated, it can be evaluated。

Finally, I have another question. My training has been completed, but when I want to view Loss in tensorboard, the following error appears. How can I solve it?