open-mmlab/OpenPCDet #1201: Unable to run Demo with CADDN

Mi97ch, 2022-11-28:
I'm running demo.py with this line:
python demo.py --cfg_file cfgs/kitti_models/CaDDN.yaml --ckpt caddn_pcdet.pth --data_path ../data/kitti/training/velodyne/000008.bin

I get this error:
Traceback (most recent call last):
  File "demo.py", line 112, in <module>
    main()
  File "demo.py", line 94, in main
    for idx, data_dict in enumerate(demo_dataset):
  File "demo.py", line 59, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)
  File "/content/gdrive/My Drive/Project/OpenPCDet/pcdet/datasets/dataset.py", line 183, in prepare_data
    data_dict=data_dict
  File "/content/gdrive/My Drive/Project/OpenPCDet/pcdet/datasets/processor/data_processor.py", line 210, in forward
    data_dict = cur_processor(data_dict=data_dict)
  File "/content/gdrive/My Drive/Project/OpenPCDet/pcdet/datasets/processor/data_processor.py", line 192, in downsample_depth_map
    image=data_dict['depth_maps'],
KeyError: 'depth_maps'

I've generated correctly the pkl files.
I've seen that the issue might be solved by modifying the demo.py code, but I didn't understand how to do that.
Do someone have any suggestion on how to do that?

Thanks for your time

Gang-Chen-China, 2022-11-29:
this means you dont‘t have depth_maps，https://github.com/open-mmlab/OpenPCDet/blob/master/docs/GETTING_STARTED.md

Mi97ch, 2022-11-29:
Hi @Gang-Chen-China, thanks for the reply.
I actually did that, running this line:
`! python -m pcdet.datasets.kitti.kitti_dataset create_kitti_infos tools/cfgs/dataset_configs/kitti_dataset.yaml`

At the end it says
 `---------------Data preparation Done---------------`
so I think it might have been successful and this is not the problem.

The error is still there

closed 2023-01-13 by github-actions[bot]

vasileiat, 2023-04-20:
i get the same error did you solve this?