open-mmlab/OpenPCDet #1202: Uable to test on nuScenes Dataset

Townjj, 2022-11-29:
Hello,
Thanks for your attention.
I try to test my model in nuScenes dataset. But it does not work in my config.
First, i successfully generate the test pkl file.
```python
-data
    |- nuscenes
            |- v1.0-trainval
            |- v1.0-test
                    |-gt_database_10sweeps_withvelo(empty), maps, samples, sweeps, v1.0-test, nuscenes_dbinfos_10sweeps_withvelo.pkl, nuscenes_infos_10sweeps_test.pkl
```
Secondly, as i change to kitti dataset config, i changed the nuscenes_dataset.yaml as follows.
```python
# nuscenes_dataset.yaml
VERSION: 'v1.0-test', # v1.0-trainval
DATA_SPLIT: {
    'train': train,
    'test': test # val
},
INFO_PATH: {
    'train': [nuscenes_infos_10sweeps_train.pkl],
    'test': [nuscenes_infos_10sweeps_test.pkl], # nuscenes_infos_10sweeps_val.pkl
}
```
I don't knew whether the **v1.0-test path** or the **setting in nuscenes.yaml problem**. Can you tell me how to change the settings in nuScenes_dataset.yaml to generate the prediction in nuscnenes test set?
thank you.

yukang2017, 2022-11-29:
Hi, what is the error message with this setting?

Townjj, 2022-11-29:
The complete yaml file:
```python
DATASET: 'NuScenesDataset'
DATA_PATH: '/PCDet/data/nuscenes'

VERSION: 'v1.0-test' #'v1.0-test' #'v1.0-trainval'
MAX_SWEEPS: 10
PRED_VELOCITY: True
SET_NAN_VELOCITY_TO_ZEROS: True
FILTER_MIN_POINTS_IN_GT: 1

DATA_SPLIT: {
    'train': train,
    'test': test #test val
}

INFO_PATH: {
    'train': [nuscenes_infos_10sweeps_train.pkl],
    'test': [nuscenes_infos_10sweeps_test.pkl], # nuscenes_infos_10sweeps_test.pkl nuscenes_infos_10sweeps_val.pkl
}

POINT_CLOUD_RANGE: [-51.2, -51.2, -5.0, 51.2, 51.2, 3.0]

BALANCED_RESAMPLING: True 

DATA_AUGMENTOR:
    DISABLE_AUG_LIST: ['placeholder']
    AUG_CONFIG_LIST:
        - NAME: gt_sampling
          DB_INFO_PATH:
              - /PCDet/data/nuscenes/v1.0-trainval/nuscenes_dbinfos_10sweeps_withvelo.pkl
          PREPARE: {
             filter_by_min_points: [
                 'car:5','truck:5', 'construction_vehicle:5', 'bus:5', 'trailer:5',
                 'barrier:5', 'motorcycle:5', 'bicycle:5', 'pedestrian:5', 'traffic_cone:5'
             ],
          }

          SAMPLE_GROUPS: [
              'car:2','truck:3', 'construction_vehicle:7', 'bus:4', 'trailer:6',
              'barrier:2', 'motorcycle:6', 'bicycle:6', 'pedestrian:2', 'traffic_cone:2'
          ]

          NUM_POINT_FEATURES: 5
          DATABASE_WITH_FAKELIDAR: False
          REMOVE_EXTRA_WIDTH: [0.0, 0.0, 0.0]
          LIMIT_WHOLE_SCENE: True

        - NAME: random_world_flip
          ALONG_AXIS_LIST: ['x', 'y']

        - NAME: random_world_rotation
          WORLD_ROT_ANGLE: [-0.3925, 0.3925]

        - NAME: random_world_scaling
          WORLD_SCALE_RANGE: [0.95, 1.05]

POINT_FEATURE_ENCODING: {
    encoding_type: absolute_coordinates_encoding,
    used_feature_list: ['x', 'y', 'z', 'intensity', 'timestamp'],
    src_feature_list: ['x', 'y', 'z', 'intensity', 'timestamp'],
}

DATA_PROCESSOR:
    - NAME: mask_points_and_boxes_outside_range
      REMOVE_OUTSIDE_BOXES: True

    - NAME: shuffle_points
      SHUFFLE_ENABLED: {
        'train': True,
        'test': True
      }

    - NAME: transform_points_to_voxels
      VOXEL_SIZE: [0.1, 0.1, 0.2]
      MAX_POINTS_PER_VOXEL: 10
      MAX_NUMBER_OF_VOXELS: {
        'train': 60000,
        'test': 60000
      }
```

and the error:
``` bash
Traceback (most recent call last):
  File "test.py", line 203, in <module>
    main()
  File "test.py", line 199, in main
    eval_single_ckpt(model, test_loader, args, eval_output_dir, logger, epoch_id, dist_test=dist_test)
  File "test.py", line 67, in eval_single_ckpt
    result_dir=eval_output_dir, save_to_file=args.save_to_file
  File "/PCDet/PointpillarV8_/tools/eval_utils/eval_utils.py", line 56, in eval_one_epoch
    for i, batch_dict in enumerate(dataloader):
  File "/usr/local/lib/python3.7/dist-packages/torch/utils/data/dataloader.py", line 517, in __next__
    data = self._next_data()
  File "/usr/local/lib/python3.7/dist-packages/torch/utils/data/dataloader.py", line 1199, in _next_data
    return self._process_data(data)
  File "/usr/local/lib/python3.7/dist-packages/torch/utils/data/dataloader.py", line 1225, in _process_data
    data.reraise()
  File "/usr/local/lib/python3.7/dist-packages/torch/_utils.py", line 429, in reraise
    raise self.exc_type(msg)
KeyError: Caught KeyError in DataLoader worker process 0.
Original Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/torch/utils/data/_utils/worker.py", line 202, in _worker_loop
    data = fetcher.fetch(index)
  File "/usr/local/lib/python3.7/dist-packages/torch/utils/data/_utils/fetch.py", line 44, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/usr/local/lib/python3.7/dist-packages/torch/utils/data/_utils/fetch.py", line 44, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "../pcdet/datasets/nuscenes/nuscenes_dataset.py", line 144, in __getitem__
    gt_boxes = data_dict['gt_boxes']
KeyError: 'gt_boxes'
```

yukang2017, 2022-11-29:
Would you please try to change [this line](https://github.com/open-mmlab/OpenPCDet/blob/028ed72eb5df413cf33ca7091b1a977495cc700c/pcdet/datasets/nuscenes/nuscenes_dataset.py#L143) to 

        `if self.dataset_cfg.get('SET_NAN_VELOCITY_TO_ZEROS', False) and 'gt_boxes' in data_dict:`

Townjj, 2022-11-30:
Thank you yukang, 
we solved this problem by the solution. I update the comment for valid information about this problem. And i think we can optimize the code in PCDet.

closed 2022-11-30 by Townjj

yukang2017, 2022-11-30:
Thanks. 

I have pushed this to it. It should be solved with the updated code.

zbaishancha, 2023-09-07:
@Townjj hello，I want to train a model using the nuscenes dataset and then test that model against the TEST dataset and submit the results of the test to the official nuscenes website, can you please give me some help？