open-mmlab/OpenPCDet #1205: Release of MPPNet checkpoint

kcyu2014, 2022-11-30:
Hi,

Thanks for the amazing work and open-source the MPPNet! 

In the readme file of MPPNet, there is a section of evaluating the pretrained model. However, I have checked several places but haven't found the download link for this pretrained model. I am not sure if I miss something, but is there any chance that you could provide a pretrained model for quick evaluation?

Best regards,
Kaicheng

jihanyang, 2022-12-02:
If you are asking for the waymo pretrained models, we cannot release them publicly according to Waymo license. You can directly email to xuesong @Cedarch for the pretrained model for academic purpose.

Cedarch, 2022-12-06:
Hi kcyu2014, sorry for late reply. The eamil is correct and I will reorganize the ckpt to you as soon as possible (the released code is reorganized and my original ckpt can not be directly loaded).

kcyu2014, 2022-12-07:
Thanks for the update! Sure I can wait until it's ready.

Cedarch, 2023-01-02:
Hi kcyu2014,  if you don't receive the model of MPPNet, you can email to me (chenxuesong@link.cuhk.edu.hk). I am not sure whether I have sent the link to  you  :).

closed 2023-02-03 by kcyu2014