open-mmlab/OpenPCDet #1212: pcdet installation as part of Dockerfile

vikasnr, 2022-12-05:
I want to run OpenPCDet in a docker container without interactive mode to do continuous bulk inference. See below my Dockerfile.

```
FROM nvidia/cuda:11.3.0-cudnn8-devel-ubuntu18.04

RUN echo 'debconf debconf/frontend select Noninteractive' | debconf-set-selections

# Install basics
RUN apt-get update -y \
    && apt-get install build-essential \
    && apt-get install -y apt-utils git curl ca-certificates bzip2 tree htop wget \
    && apt-get install -y libglib2.0-0 libsm6 libxext6 libxrender-dev bmon iotop g++ python3.7 python3.7-dev python3.7-distutils

# Install cmake v3.13.2
RUN apt-get purge -y cmake && \
    mkdir /root/temp && \
    cd /root/temp && \
    wget https://github.com/Kitware/CMake/releases/download/v3.13.2/cmake-3.13.2.tar.gz && \
    tar -xzvf cmake-3.13.2.tar.gz && \
    cd cmake-3.13.2 && \
    bash ./bootstrap && \
    make && \
    make install && \
    cmake --version && \
    rm -rf /root/temp

# Install python
RUN ln -sv /usr/bin/python3.7 /usr/bin/python
RUN wget https://bootstrap.pypa.io/get-pip.py && \
	python get-pip.py && \
	rm get-pip.py

# Install python packages
RUN PIP_INSTALL="python -m pip --no-cache-dir install" && \
    $PIP_INSTALL numpy==1.19.3 llvmlite numba 

# Install torch and torchvision
# See https://pytorch.org/ for other options if you use a different version of CUDA
RUN pip3 install torch==1.10.2+cu113 torchvision==0.11.3+cu113 torchaudio==0.10.2+cu113 -f https://download.pytorch.org/whl/cu113/torch_stable.html

RUN pip3 install numpy==1.20 #to solve numpy dependency issues with opencv!

# Install python packages
RUN PIP_INSTALL="python -m pip --no-cache-dir install" && \
    $PIP_INSTALL tensorboardX easydict pyyaml scikit-image tqdm SharedArray six

WORKDIR /root

# Install Boost geometry
RUN wget https://jaist.dl.sourceforge.net/project/boost/boost/1.68.0/boost_1_68_0.tar.gz && \
    tar xzvf boost_1_68_0.tar.gz && \
    cp -r ./boost_1_68_0/boost /usr/include && \
    rm -rf ./boost_1_68_0 && \
    rm -rf ./boost_1_68_0.tar.gz 

# A weired problem that hasn't been solved yet
RUN pip uninstall -y SharedArray && \
    pip install SharedArray

RUN pip install spconv-cu113

#just for running the Demo(without visualization) we need to add:
RUN apt-get update && apt-get install -y python3-opencv
RUN pip install opencv-python
RUN pip install open3d

#to permanently export the location of OpenPCdet (installed in a persistent volume) to PYTHONPATH 
ENV PYTHONPATH="/root/data/OpenPCDet/:$PYTHONPATH"

# copy the contents from local to docker 
[... 134 lines cut ...]
    _build_ext.run(self)
  File "/usr/local/lib/python3.7/dist-packages/setuptools/_distutils/command/build_ext.py", line 346, in run
    self.build_extensions()
  File "/usr/local/lib/python3.7/dist-packages/torch/utils/cpp_extension.py", line 735, in build_extensions
    build_ext.build_extensions(self)
  File "/usr/local/lib/python3.7/dist-packages/setuptools/_distutils/command/build_ext.py", line 468, in build_extensions
    self._build_extensions_serial()
  File "/usr/local/lib/python3.7/dist-packages/setuptools/_distutils/command/build_ext.py", line 494, in _build_extensions_serial
    self.build_extension(ext)
  File "/usr/local/lib/python3.7/dist-packages/setuptools/command/build_ext.py", line 246, in build_extension
    _build_ext.build_extension(self, ext)
  File "/usr/local/lib/python3.7/dist-packages/setuptools/_distutils/command/build_ext.py", line 556, in build_extension
    depends=ext.depends,
  File "/usr/local/lib/python3.7/dist-packages/torch/utils/cpp_extension.py", line 551, in unix_wrap_ninja_compile
    cuda_post_cflags = unix_cuda_flags(cuda_post_cflags)
  File "/usr/local/lib/python3.7/dist-packages/torch/utils/cpp_extension.py", line 450, in unix_cuda_flags
    cflags + _get_cuda_arch_flags(cflags))
  File "/usr/local/lib/python3.7/dist-packages/torch/utils/cpp_extension.py", line 1606, in _get_cuda_arch_flags
    arch_list[-1] += '+PTX'
IndexError: list index out of range

```
Although I'm running this on a GPU machine error stack starts with `No CUDA runtime is found, using CUDA_HOME='/usr/local/cuda'.`

Is the error related to docker build process not being able to use GPU?

vikasnr, 2022-12-07:
Resolved it by installing pcdet at docker runtime rather than installing while docker build.

closed 2022-12-07 by vikasnr

hurileborjigin, 2024-12-12:
> Resolved it by installing pcdet at docker runtime rather than installing while docker build.

Hi, did you find out the reason for it? I have the same issue and solved it in the same way.