open-mmlab/OpenPCDet #1214: Unable to run demo.py inference on custom data

purple-mesh, 2022-12-07:
Hi All,

I have been trying to get the `demo.py` inference example to run on point cloud data which I have collected myself.

I saw in a few other GitHub issues that mentioned the `POINT_CLOUD_RANGE` should be updated, but I only change one of the parameters, I get the following.

```console
$ python demo.py --cfg_file cfgs/custom_models/pv_rcnn.yaml \
    --ckpt pv_rcnn_8369.pth \
    --data_path my_data.npy

[skipped for brevity ... ]

2022-12-07 16:50:55,413   INFO  ==> Done (loaded 277/319)
2022-12-07 16:50:56,642   INFO  Visualized sample index: 	1
Traceback (most recent call last):
  File "demo.py", line 132, in <module>
    main()
  File "demo.py", line 106, in main
    pred_dicts, _ = model.forward(data_dict)
  File "/work/temp/OpenPCDet/pcdet/models/detectors/pv_rcnn.py", line 11, in forward
    batch_dict = cur_module(batch_dict)
  File "/root/.local/lib/python3.7/site-packages/torch/nn/modules/module.py", line 722, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/work/temp/OpenPCDet/pcdet/models/backbones_3d/pfe/voxel_set_abstraction.py", line 407, in forward
    point_features = self.vsa_point_feature_fusion(point_features.view(-1, point_features.shape[-1]))
  File "/root/.local/lib/python3.7/site-packages/torch/nn/modules/module.py", line 722, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/root/.local/lib/python3.7/site-packages/torch/nn/modules/container.py", line 117, in forward
    input = module(input)
  File "/root/.local/lib/python3.7/site-packages/torch/nn/modules/module.py", line 722, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/root/.local/lib/python3.7/site-packages/torch/nn/modules/linear.py", line 91, in forward
    return F.linear(input, self.weight, self.bias)
  File "/root/.local/lib/python3.7/site-packages/torch/nn/functional.py", line 1676, in linear
    output = input.matmul(weight.t())
RuntimeError: mat1 dim 1 must match mat2 dim 0
```

Other times, the inference completes, but there is no prediction output and the visualiser is empty, so I *think* it could have been incorrect dataset/model configuration.

---

My setup and what I have done so far:

* Used the official docker script for development environment
* Ensured that `demo.py` works as expected using the velodyne kitti data set
* Sanity check the custom data in `my_data.npy` by using the following

```python
import open3d
from visual_utils import open3d_vis_utils as V

NPY_PATH = "/path/to/my_data.npy"
data = np.load(NPY_PATH)
V.draw_scenes(points=data)
```

For reference, I have attached one frame of my custom point cloud data below as a `.zip` file which can be decompressed into a `.npy` file.

[image]
[my_data.zip](https://github.com/open-mmlab/OpenPCDet/files/10180691/my_data.zip)

I am still very new at exploring point cloud object detection, so any advice regarding on how to tweak the `tools/cfgs/dataset_configs/custom_dataset.yaml` and `tools/cfgs/custom_models/pv_rcnn.yaml` would be appreciated.

jihanyang, 2022-12-08:
Hello, you need to modify the DemoDataset to fit your custom dataset. Nevertheless, I recommand you to add visualization code in eval_utils as demo.py does (I also use this solution).

takeyamayuki, 2023-01-10:
Hi, I am also trying to inference on a custom data `npy` file, but it is not drawing properly in `demo.py`.

I have converted the kittidataset `bin` file to `npy` with the following code and input it into `demo.py`, but it does not render properly.
```python
import numpy as np
binFile = "/home/yonelab/workspace/lidar_data/data_object_velodyne/training/velodyne/000008.bin"
OUTPUT = "/home/yonelab/workspace/lidar_data/data_object_velodyne/training/velodyne/npy/"

data = np.fromfile(binFile, dtype=np.float32).reshape(
    (-1, 4)).astype(np.float64)
# print(data.shape)
# print("[x, y, z, intensity] =", data[0])

np.save(OUTPUT+'my_pc.npy', data)
```

Since the readme says it can be run in `npy`, I think we need to support the `npy` data format as soon as possible.
Or, how to modify the `DemoDataset` to fit this custom dataset?

takeyamayuki, 2023-01-12:
I was able to generate bin from ply with the following code. I was then able to load it into `demo.py`.

```python
# cf) https://github.com/open-mmlab/mmdetection3d/blob/master/docs/en/getting_started.md

import numpy as np
import pandas as pd
from plyfile import PlyData

def convert_ply(input_path, output_path):
    plydata = PlyData.read(input_path)  # read file
    data = plydata.elements[0].data  # read data
    # print(data)
    data_pd = pd.DataFrame(data)  # convert to DataFrame
    # initialize array to store data
    data_np = np.zeros(data_pd.shape, dtype=np.float32)
    property_names = data[0].dtype.names  # read names of properties
    for i, name in enumerate(
            property_names):  # read data by property
        data_np[:, i] = data_pd[name]
    new_col = np.zeros((data_np.shape[0], 1))
    data_np = np.append(data_np, new_col, axis=1)
    print(data_np)
    data_np.astype(np.float32).tofile(output_path)

convert_ply('/home/yonelab/workspace/lidar_data/DATA_AOYAMA-suiyoubi/ply/test_data.ply',
            '/home/yonelab/workspace/lidar_data/DATA_AOYAMA-suiyoubi/bin/test.bin')
```

closed 2023-02-26 by github-actions[bot]

reynoldms, 2024-07-12:
If you wanna do the demo to `.npy` format, just add `--ext '.npy'` when running the `demo.py`
For example,
`$ python demo.py --cfg_file cfgs/custom_models/pv_rcnn.yaml --ckpt pv_rcnn_8369.pth --data_path my_data.npy --ext '.npy'`
  
The `demo.py` use `.bin` format as default extension of point cloud file to show.