open-mmlab/OpenPCDet #1215: error when training on multiple  gpus

WYHbxer, 2022-12-08:
I am trying to train the net on multiple gpus. here is my code 
`import argparse
import datetime
import glob
import os

print(os.getcwd())#显示当前路径
  
os.chdir('/tmp/OpenPCDet/tools')#更改路径，''里面为更改的路径
  
print(os.getcwd())#显示当前路径

from pathlib import Path
from test import repeat_eval_ckpt

import torch
import torch.distributed as dist
import torch.nn as nn
from tensorboardX import SummaryWriter

from pcdet.config import cfg, cfg_from_list, cfg_from_yaml_file, log_config_to_file
from pcdet.datasets import build_dataloader
from pcdet.models import build_network, model_fn_decorator
from pcdet.utils import common_utils
from train_utils.optimization import build_optimizer, build_scheduler
from train_utils.train_utils import train_model
os.environ["CUDA_VISIBLE_DEVICES"] = "0,1,2,3" 

def parse_config():
    parser = argparse.ArgumentParser(description='arg parser')
    parser.add_argument('--cfg_file', type=str, default=None, help='specify the config for training')

    parser.add_argument('--batch_size', type=int, default=None, required=False, help='batch size for training')
    parser.add_argument('--epochs', type=int, default=None, required=False, help='number of epochs to train for')
    parser.add_argument('--workers', type=int, default=8, help='number of workers for dataloader')
    parser.add_argument('--extra_tag', type=str, default='default', help='extra tag for this experiment')
    parser.add_argument('--ckpt', type=str, default=None, help='checkpoint to start from')
    parser.add_argument('--pretrained_model', type=str, default=None, help='pretrained_model')
    parser.add_argument('--launcher', choices=['none', 'pytorch', 'slurm'], default='none')
    parser.add_argument('--tcp_port', type=int, default=18888, help='tcp port for distrbuted training')
    parser.add_argument('--sync_bn', action='store_true', default=False, help='whether to use sync bn')
    parser.add_argument('--fix_random_seed', action='store_true', default=False, help='')
    parser.add_argument('--ckpt_save_interval', type=int, default=1, help='number of training epochs')
    parser.add_argument('--local_rank', type=int, default=0, help='local rank for distributed training')
    parser.add_argument('--max_ckpt_save_num', type=int, default=30, help='max number of saved checkpoint')
    parser.add_argument('--merge_all_iters_to_one_epoch', action='store_true', default=False, help='')
    parser.add_argument('--train_info', type=str, default="None", help='train info file')
    parser.add_argument('--set', dest='set_cfgs', default=None, nargs=argparse.REMAINDER,
                        help='set extra config keys if needed')

    parser.add_argument('--max_waiting_mins', type=int, default=0, help='max waiting minutes')
    parser.add_argument('--start_epoch', type=int, default=0, help='')
    parser.add_argument('--save_to_file', action='store_true', default=False, help='')
    parser.add_argument('--lr', type=float, default=0.004,  help='强行给yaml中的文件参数给换了, 方便在外面改')

    args = parser.parse_args()

    cfg_from_yaml_file(args.cfg_file, cfg)
    cfg.TAG = Path(args.cfg_file).stem
    cfg.EXP_GROUP_PATH = '/'.join(args.cfg_file.split('/')[1:-1])  # remove 'cfgs' and 'xxxx.yaml'

    if args.set_cfgs is not None:
        cfg_from_list(args.set_cfgs, cfg)

    return args, cfg

def main():
    args, cfg = parse_config()
    if args.launcher == 'none':
        dist_train = False
[... 243 lines cut ...]
  msg: "Process failed with exitcode 1"
=======================================
Other Failures:
[1]:
  time: 2022-12-08_09:37:35
  rank: 1 (local_rank: 1)
  exitcode: 1 (pid: 65932)
  error_file: <N/A>
  msg: "Process failed with exitcode 1"
[2]:
  time: 2022-12-08_09:37:35
  rank: 2 (local_rank: 2)
  exitcode: 1 (pid: 65933)
  error_file: <N/A>
  msg: "Process failed with exitcode 1"
***************************************`

I am using
GPU: GTX 3090 
torch: 1.9.0+cu111
spconv: spconv-cu113           2.1.25

I can run the code on a sinlge gpu

help me !

closed 2023-01-23 by github-actions[bot]

d33dler, 2023-08-28:
Did you find a way to fix this? @YanhaoWu

YanhaoWu, 2023-08-28:
> Did you find a way to fix this? @YanhaoWu

No, I have no way