open-mmlab/OpenPCDet #1216: Too large RAM used with train centerpoint_4frames

MaxShi007, 2022-12-09:
I want to reproduct  the MPPnet, so i following the [guideline](https://github.com/open-mmlab/OpenPCDet/blob/master/docs/guidelines_of_approaches/mppnet.md)。
But I used 200G RAM with 2 V100 (32G) when i training cneterpoint_4frames。it s too expensive because per epoch use 20 hour. I have more GPU but not have so much RAM.
Is it normal to use so much RAM？
How to reduce RAM usage?
For other friend， If you successfully train centerpoint_4frames, I want to know how you train

jihanyang, 2022-12-10:
May I know your numba version?

MaxShi007, 2022-12-11:
> 

numba== 0.51.2

MaxShi007, 2022-12-13:
I found that the dataloader iteration was used twice [here ](https://github.com/open-mmlab/OpenPCDet/blob/4713332c5b73b32ac23b425022a06861bfa23b89/tools/train_utils/train_utils.py#L16)and [here](https://github.com/open-mmlab/OpenPCDet/blob/4713332c5b73b32ac23b425022a06861bfa23b89/tools/train_utils/train_utils.py#L145)，This results in double RAM used

I think  [here ](https://github.com/open-mmlab/OpenPCDet/blob/4713332c5b73b32ac23b425022a06861bfa23b89/tools/train_utils/train_utils.py#L15) shoule be "total_it_each_epoch != len(train_loader)"
Am i right?

jihanyang, 2022-12-14:
Sorry for late reply. Would you willing to try numba==0.48.0? It seems that 0.51.2 will cause extra RAM usage in our recent obsveration.

MaxShi007, 2022-12-15:
> 

I have tried，but its not work

Cedarch, 2022-12-22:
Hi @MaxShi007， you can reduce the num_workers to save RAM cost but it will cost more training time. Besides, if you only focus on MPPNet,  I can  share our pretrained ckpt to you. Here is my email: chenxuesong@link.cuhk.edu.hk

closed 2023-02-05 by github-actions[bot]

HaoranZhuExplorer, 2024-11-25:
> numba==0.48.0

thank you. this is very useful. Is it possible you explicitly specify each package's version in requirements.txt? It turns out that it's very difficult to set up the environment correctly...