open-mmlab/OpenPCDet #1217: A100 env setting

csj3684, 2022-12-13:
Hi, I really appreciate your works.

I used this model with 
Ubuntu 18.04, Driver Version: 470.161.03, Titan RTX, python 3.6.13, Pytorch 1.7.1, Cuda 10.2, Cudnn 8.2.4 ( for Cuda 10.2 )
and It has been working.

And I need to do with 
Ubuntu 20.04,  Driver Version: 525.60.11, A100 (80GB), Cuda 11+, Cudnn 8.6.0 ( for Cuda 11.x )

I tried all combinations from

python 3.6.13 ~ python 3.9.12
torch 1.8.0 ~ 1.10.0
Cuda 11.1

but nothing works ( touch.cuda.is_available() return False always )
all installation is done without any error and all env PATH properly setted.
I cannot use this on Cuda 11+ ?? T . T

csj3684, 2022-12-19:
> I use pcdet with CUDA 11.1 + pytorch 1.8.0 + python3.7 on A100 80G.

Thank you for replying Is it Ubuntu 20.04 ? 18.04 ?

jihanyang, 2022-12-19:
This is a cluster which uses centos 7.

csj3684, 2022-12-19:
> This is a cluster which uses centos 7.

Thank you !

closed 2023-02-03 by github-actions[bot]

kbratsy, 2023-10-04:
> I use pcdet with CUDA 11.1 + pytorch 1.8.0 + python3.7 on A100 80G.

Hello,
Which spconv version did you use? How did you install it?