open-mmlab/OpenPCDet #1221: Why execute the same code for the same frame of data but get different detection results?

wangxiyang2022, 2022-12-19:
Thank you for your great work in making the OpenPCDet project available for 3D object detection.

When I used PointRCNN to test the same frame of data, I found that the detection results obtained in each run were different.

For example, the detection results I got when I executed the code for the first time are as follows:

[image]

But when I re-execute the code, I get the following detection results:

[image]

Why execute the same code for the same frame of data but get different detection results?

jihanyang, 2022-12-19:
May I know the command you execute?

wangxiyang2022, 2022-12-19:
Thanks for your reply, here is my command：

`python demo.py --cfg_file cfgs/kitti_models/pointrcnn.yaml --ckpt pointrcnn_7870.pth --data_path /home/wangxiyang/Desktop/code_test/DetMOT/data/KITTI/tracking/train/velodyne/0000/000000.bin
`

jihanyang, 2022-12-19:
please try to fix the random seed of numpy in demo.py as test.py does.

closed 2022-12-19 by jihanyang

wangxiyang2022, 2022-12-20:
I found that a fixed number of point clouds were randomly selected during data preprocessing.

[image]

 Since the point clouds selected each time are random, the final detection results will also be different. Why is it needed in the testing phase?