open-mmlab/OpenPCDet #1225: Custom dataset evaluation result is abnormal

S-kewen, 2022-12-30:
First of all, I would like to thank the author for his contribution. We can easily use multiple frameworks for training and verification.

We recently ran into some issues and didn't see how to fix them. We try to label our dataset (KITTI-format) in Carla simulator for evaluation, but we found that the AP results are all 0.00 in the all models. 

We think this should not be the dataset bug because we have tried to check it with some KITTI-visualization tools. Also, we use a separate PointRCNN repository ([sshaoshuai](https://github.com/sshaoshuai)
/
[PointRCNN](https://github.com/sshaoshuai/PointRCNN)) for evaluation and everything works fine.

Thank you very much!

jrb-cobot, 2023-05-30:
Did you ever solve this?

AJAY31797, 2024-02-11:
Hey @S-kewen,

Could you solve the issue? I am also getting the AP value for all classes as 0 in evaluation. 

Best Regards,
Ajay