open-mmlab/OpenPCDet #1230: Domain Adaptation (Trained PVRCNN on Waymo, Testing on Kitti), Error: Not Updated Weight, Poor Performance

AbhishekKaushikCV, 2023-01-05:
Hey Guys,
- I have trained the PVRCNN on Waymo with the given Config file from PCDET. 
- The Waymo dataset Config, that I used for training the above model, for features are:

`POINT_FEATURE_ENCODING: {
    encoding_type: absolute_coordinates_encoding,
    used_feature_list: ['x', 'y', 'z', 'intensity', 'elongation'],
    src_feature_list: ['x', 'y', 'z', 'intensity', 'elongation'],
}`

Now, I am testing the performance of the above trained PVRCNN model (with the exact config as it was in Training) on the KITTI dataset.  **Waymo -> Kitti**

### Setting 1.
- The source dataset config (which is **Waymo**) are:

`POINT_FEATURE_ENCODING: {
    encoding_type: absolute_coordinates_encoding,
    used_feature_list: ['x', 'y', 'z', 'intensity', 'elongation'],
    src_feature_list: ['x', 'y', 'z', 'intensity', 'elongation'],
}`

- The target dataset config (which is **Kitti**) are:

`POINT_FEATURE_ENCODING: {
    encoding_type: absolute_coordinates_encoding,
    used_feature_list: ['x', 'y', 'z', 'intensity','timestamp'],
    src_feature_list: ['x', 'y', 'z', 'intensity','timestamp'],
}`

**ERROR 1.** with the **Setting1.** (**all the pretrained weights are updating successfully**), I am getting this error (error snippet):

`  File "../pcdet/datasets/processor/data_processor.py", line 134, in transform_points_to_voxels
    voxel_output = self.voxel_generator.generate(points)
  File "../pcdet/datasets/processor/data_processor.py", line 54, in generate
    voxel_output = self._voxel_generator.point_to_voxel(tv.from_numpy(points))
RuntimeError: /io/build/temp.linux-x86_64-cpython-37/spconv/build/src/csrc/sparse/all/ops_cpu3d/Point2VoxelCPU/Point2VoxelCPU_point_to_voxel_static.cc(22)
num_features == voxels.dim(2) assert faild. your points num features doesn't equal to voxel.
`
- **evaluation is not working because of the above error**

### Setting 2. 

- The source dataset config (which is **Waymo**) are:

`POINT_FEATURE_ENCODING: {
    encoding_type: absolute_coordinates_encoding,
    used_feature_list: ['x', 'y', 'z', 'intensity', 'elongation'],
    src_feature_list: ['x', 'y', 'z', 'intensity', 'elongation'],
}`

- The target dataset config (which is **Kitti**) are:

`POINT_FEATURE_ENCODING: {
    encoding_type: absolute_coordinates_encoding,
    used_feature_list: ['x', 'y', 'z', 'intensity'],
    src_feature_list: ['x', 'y', 'z', 'intensity','timestamp'],
}`

**NO ERROR (But performance is very low).** with Setting2. (**all the pretrained weights are not updating successfully**).
- weights which are  not updating are:

`INFO  Not updated weight backbone_3d.conv_input.0.weight: torch.Size([16, 3, 3, 3, 4])
INFO  Not updated weight pfe.SA_rawpoints.mlps.0.0.weight: torch.Size([16, 4, 1, 1])
INFO  Not updated weight pfe.SA_rawpoints.mlps.1.0.weight: torch.Size([16, 4, 1, 1])
 INFO  ==> Done (loaded 316/319)`

- **evaluation is working without any error but getting very poor performance**
### Questions:
- Is there any way to solve this problem (load all the weights, and also complete the evaluation run) without training a new model (with changed feature list)?
- What other things should I keep in my mind (if I want to train a model for testing on different datasets),  while testing the performance on different dataset?

Thanks!!

jihanyang, 2023-01-06:
There are a lot of configs need to be modified for domain adaptation between waymo and kitti. May I know have you checked ST3D? It is a paper that has tried this setting based on PCDet.

AbhishekKaushikCV, 2023-01-06:
Yes, I am using ST3D as a reference, and changed the code, and config according to the ST3D.

- I am thinking, that the reason for all weights are not updating is the different features in `used_feature_list` in training and testing. 
- In ST3D paper, the dataset setting for Waymo is:

`POINT_FEATURE_ENCODING: {
    encoding_type: absolute_coordinates_encoding,
    used_feature_list: ['x', 'y', 'z'],
    src_feature_list: ['x', 'y', 'z', 'intensity', 'elongation'],
}`

- In ST3D paper, the dataset setting for Kitti is:

`POINT_FEATURE_ENCODING: {
    encoding_type: absolute_coordinates_encoding,
    used_feature_list: ['x', 'y', 'z'],
    src_feature_list: ['x', 'y', 'z', 'intensity'],
}`

jihanyang, 2023-01-07:
Hello, I think you the cfg that you checked is not extactly used. In ST3D, the used waymo config for adaptation should be the following one: 
https://github.com/CVMI-Lab/ST3D/blob/cf5ece6e43fc7c5801532847d8e96aded6adf202/tools/cfgs/dataset_configs/da_waymo_dataset.yaml#L43-L47

closed 2023-02-21 by github-actions[bot]