open-mmlab/OpenPCDet #1236: custom dataset training with more than 3 classes

sangjun7997, 2023-01-13:
I want to training with my custom dataset which have about 10 classes..
But I have understood that custom dataset training tutorial from OpenPCDet is not support to add other classes such as trafficlight or policeCar ...
So I want to know how can I use these classes to train model

jihanyang, 2023-01-13:
Actually, you can add any classes that you want in custom dataset. However, the only problem is you need to modified the KITTI evaluation to evaluate on 10 classes or write your own version. The custom dataset itself shouldn't have any constraint for the number of classes.

sangjun7997, 2023-01-13:
> 

So, I want to know how to do that..

SatoshiSolution, 2023-01-17:
Hi, 

I had that problem too, but I will not spend too much time on explaining to you, it should be sufficient to understand what to change in my code changes:

in eval.py
```python
def get_official_eval_result(gt_annos, dt_annos, current_classes, PR_detail_dict=None):
    """
    metric: eval type. 0: bbox, 1: bev, 2: 3d
    min_overlaps: float, min overlap. format: [num_overlap, metric, class]. 
    """
    # name classes           car  ped  cyc  van  per  tru  mast
    # int classes:            0    1    2    3    4    5    6 
                    #bbox     x    x    x                          
                    #bev      x    x    x                          
                    #3d       x    x    x                          
    overlap_0_7 = np.array([[0.7, 0.7, 0.7],    
                            [0.7, 0.7, 0.7],    
                            [0.7, 0.7, 0.7]])   
    
    overlap_0_5 = np.array([[0.5, 0.5, 0.5],
                            [0.5, 0.5, 0.5],
                            [0.5, 0.5, 0.5]])
    min_overlaps = np.stack([overlap_0_5, overlap_0_7], axis=0)  # [2, 3, 5]
    #change here
    class_to_name = {
        0: 'type1', #add to overlap_0_7 (also to clean_data())
        1: 'type2',
        2: 'type3'
    } 
  
```
and here:
```python
def clean_data(gt_anno, dt_anno, current_class, difficulty):
    CLASS_NAMES = ['type1', 'type2', 'type3']   #change here
    MIN_HEIGHT = [40, 25, 25]
    MAX_OCCLUSION = [0, 1, 2]
    MAX_TRUNCATION = [0.15, 0.3, 0.5]
```

Hope that helps! 
Best

QingXIA233, 2023-02-10:
@jihanyang Hello, I also tried to use PVRCNN++ on my own dataset. I have already changed the eval scrit according to my own dataset settings. Everything is all right. However, when I trained CenterPoint-Voxel and PVRCNN++ individually for 30 epochs on my own dataset, I got surprising results. The performance of PVRCNN++ is worse than that of CP-Voxel. My dataset has 11 classes instead of 3 in WOD. I wonder which parameters of PVRCNN++ I should tune to get better result than CP-Voxel?

closed 2023-03-28 by github-actions[bot]

YouthZCC, 2024-03-18:
> @jihanyang Hello, I also tried to use PVRCNN++ on my own dataset. I have already changed the eval scrit according to my own dataset settings. Everything is all right. However, when I trained [...]

Hello, I have some problems to eval my own dataset without 2d images. Could you share me with your eval to refer?   My e-mail address is zhwyouth@163.com. Thanks a lot

Chenycan, 2025-01-13:
CLASS_NAMES: ['boat', 'pontoon', 'buoy','pier','bridge','architecture','islands','shore']

MAP_CLASS_TO_KITTI: {
    'boat': 'boat',
    'pontoon': 'pontoon',
    'buoy': 'buoy',
    'pier': 'pier',
    'bridge':'bridge',
    'architecture':'architecture',
    'islands':'islands',
    'shore':'shore'
}
DATA_CONFIG: 
    _BASE_CONFIG_: /home/guest/2TLOCAL/CYC/npy/OpenPCDet-master_1/OpenPCDet-master/tools/cfgs/dataset_configs/custom_dataset.yaml

MODEL:
    NAME: CenterPoint

    VFE:
        NAME: MeanVFE

    BACKBONE_3D:
        NAME: VoxelResBackBone8x

    MAP_TO_BEV:
        NAME: HeightCompression
        NUM_BEV_FEATURES: 256

    BACKBONE_2D:
        NAME: BaseBEVBackbone

        LAYER_NUMS: [5, 5]
        LAYER_STRIDES: [1, 2]
        NUM_FILTERS: [128, 256]
        UPSAMPLE_STRIDES: [1, 2]
        NUM_UPSAMPLE_FILTERS: [256, 256]

    DENSE_HEAD:
        NAME: CenterHead
        CLASS_AGNOSTIC: False

        CLASS_NAMES_EACH_HEAD: [
            ['boat', 'pontoon', 'buoy','pier','bridge','architecture','islands','shore']
        ]

        SHARED_CONV_CHANNEL: 64
        USE_BIAS_BEFORE_NORM: True  # TODO
        NUM_HM_CONV: 2  # TODO
        SEPARATE_HEAD_CFG:
            HEAD_ORDER: ['center', 'center_z', 'dim', 'rot']
            HEAD_DICT: {
                'center': {'out_channels': 2, 'num_conv': 2},
                'center_z': {'out_channels': 1, 'num_conv': 2},
                'dim': {'out_channels': 3, 'num_conv': 2},
                'rot': {'out_channels': 2, 'num_conv': 2},
            }

        TARGET_ASSIGNER_CONFIG:
            FEATURE_MAP_STRIDE: 8
            NUM_MAX_OBJS: 500
            GAUSSIAN_OVERLAP: 0.1
            MIN_RADIUS: 2
            DENSE_REG: 1

        LOSS_CONFIG:
            LOSS_WEIGHTS: {
                'cls_weight': 1.0,
                'loc_weight': 1.0,
                'code_weights': [1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0]
            }

        POST_PROCESSING:
            SCORE_THRESH: 0.1
            POST_CENTER_LIMIT_RANGE: [-75.2, -75.2, -5.0, 75.2, 75.2, 3.0]
            MAX_OBJ_PER_SAMPLE: 500
            NMS_CONFIG:
                MULTI_CLASSES_NMS: False
                NMS_TYPE: nms_gpu
                NMS_THRESH: 0.01
                NMS_PRE_MAXSIZE: 4096
                NMS_POST_MAXSIZE: 500
            

    POST_PROCESSING:
      RECALL_THRESH_LIST: [0.3, 0.5, 0.7]

      OUTPUT_RAW_SCORE: False

      EVAL_METRIC: once

OPTIMIZATION:
    BATCH_SIZE_PER_GPU: 4
    NUM_EPOCHS: 80

    OPTIMIZER: adam_onecycle
    LR: 0.003
    WEIGHT_DECAY: 0.01
    MOMENTUM: 0.9

    MOMS: [0.95, 0.85]
    PCT_START: 0.4
    DIV_FACTOR: 10
    DECAY_STEP_LIST: [35, 45]
    LR_DECAY: 0.1
    LR_CLIP: 0.0000001

    LR_WARMUP: False
    WARMUP_EPOCH: 1

    GRAD_NORM_CLIP: 35

以上是Yaml文件，报错RuntimeError: Given groups=1, weight of size [128, 256, 3, 3], expected input[2, 2048, 201, 242] to have 256 channels, but got 2048 channels instead，该怎么修改？