open-mmlab/OpenPCDet #1240: Extract point cloud data from tfrecord and generate data infos but showed 'Killed'

XiaoyanQian, 2023-01-15:
[image]

when run the" python -m pcdet.datasets.waymo.waymo_dataset --func create_waymo_infos \
    --cfg_file tools/cfgs/dataset_configs/waymo_dataset_multiframe.yaml", my server showed the "killed" in the pic. 

Anyone could tell me why it will be killed.

adv010, 2023-01-15:
Must be running out of memory when data is being loaded

XiaoyanQian, 2023-01-16:
> 

Thank you so much. Do you know how many memories it would be open to when loading the data?

XiaoyanQian, 2023-01-17:
> Must be running out of memory when data is being loaded

I have more than 500G left； logically, there might be enough for loading data. 

[image]

closed 2023-03-03 by github-actions[bot]

qifeng22, 2023-07-15:
hello,do you know how many memories it would be open to when loading the data?