open-mmlab/OpenPCDet #1242: Confusing: probably mismatching between feature map size and anchors generation

OrangeSodahub, 2023-01-22:
Using this config in `AnchorHeadSingle`:
```
    DENSE_HEAD:
        NAME: AnchorHeadSingle
        CLASS_AGNOSTIC: False

        USE_DIRECTION_CLASSIFIER: True
        DIR_OFFSET: 0.78539
        DIR_LIMIT_OFFSET: 0.0
        NUM_DIR_BINS: 2

        ANCHOR_GENERATOR_CONFIG: [
            {
                'class_name': 'Car',
                'anchor_sizes': [[3.9, 1.6, 1.56]],
                'anchor_rotations': [0, 1.57],
                'anchor_bottom_heights': [-1.78],
                'align_center': False,
                'feature_map_stride': 2,
                'matched_threshold': 0.6,
                'unmatched_threshold': 0.45
            },
```
Note that it set `feature_map_stride` to `2`, and this parameter is used in `AnchorGenerator`:
https://github.com/open-mmlab/OpenPCDet/blob/e948c537858d4e468a9ef50b023324f6bc96db2a/pcdet/models/dense_heads/anchor_head_template.py#L38-L44
If using `point_cloud_range=[-75.2, -75.2, -5, 75.2, 75.2, 3]` and `voxel_size=[0.4, 0.4, 8]` and we get `grid_size=[376, 376]`, so the `grid_size` in `anchor_generator.generate_anchors` will be `grid_size // feature_map_size = [188, 188]`, all the generated anchors are located in that-size mesh.
However, in `anchor_head_single.__init__()`:
https://github.com/open-mmlab/OpenPCDet/blob/e948c537858d4e468a9ef50b023324f6bc96db2a/pcdet/models/dense_heads/anchor_head_single.py#L7-L24
and `anchor_head_single.forward()`:
https://github.com/open-mmlab/OpenPCDet/blob/e948c537858d4e468a9ef50b023324f6bc96db2a/pcdet/models/dense_heads/anchor_head_single.py#L41-L45
the `spatial_feature_2d` is of shape `[batch_size, num_features, grid_size, grid_size]` and here is `[B, input_channels, 376, 376]`, so the output will be in shape of `[B, output_channels, 376, 376]` and I got this error:
```
  File "../pcdet/models/detectors/pointpillar.py", line 16, in forward
    loss, tb_dict, disp_dict = self.get_training_loss()
  File "../pcdet/models/detectors/pointpillar.py", line 29, in get_training_loss
    loss_rpn, tb_dict = self.dense_head.get_loss()
  File "../pcdet/models/dense_heads/anchor_head_template.py", line 222, in get_loss
    cls_loss, tb_dict = self.get_cls_layer_loss()
  File "../pcdet/models/dense_heads/anchor_head_template.py", line 133, in get_cls_layer_loss
    cls_loss_src = self.cls_loss_func(cls_preds, one_hot_targets, weights=cls_weights)  # [N, M]
  File "?/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1051, in _call_impl
    return forward_call(*input, **kwargs)
  File "../pcdet/utils/loss_utils.py", line 62, in forward
    pt = target * (1.0 - pred_sigmoid) + (1.0 - target) * pred_sigmoid
RuntimeError: The size of tensor a (353440) must match the size of tensor b (1413760) at non-singleton dimension 1
```
where 353440=188\*188\*10 and 1413760=376\*376\*10, and, what cls_pred do is to predict which class each anchor probably be, so I think the shape of `conv2d` should be `[B, output_channels, 188, 188]` rather than 376.
Is this a potential bug if I got nothing missed? If yes, should we add a parameter `stride=config.feature_map_stride(2)`? If no, could you explain what I've missed?

closed 2023-03-08 by github-actions[bot]

AJAY31797, 2024-01-31:
Hi @OrangeSodahub, 

Could you figure out what's the reason for the error. I am facing the same problem. 

Thanks, 
Ajay

OrangeSodahub, 2024-07-01:
@AJAY31797 Long time ago, I've forgot everything