open-mmlab/OpenPCDet #1248: How can I evaluate model based on test set of Once dataset?

Eaton2022, 2023-01-30:
When I try to evaluate model based on test set of Once dataset, the log information shows that the total number of Once dataset is 0.  That is weird.

dyzxzeng, 2023-02-12:
ONCE dataset has no annos in test split. According to ONCE official website(https://once-for-auto-driving.github.io/), the test results can be obtained from codalab. But codalab stops their old servers several months ago. Till now I have no idea where to submit result

Eaton2022, 2023-02-14:
> ONCE dataset has no annos in test split. According to ONCE official website(https://once-for-auto-driving.github.io/), the test results can be obtained from codalab. But codalab stops their old servers several months ago. Till now I have no idea where to submit result

Thanks for your kindly reply.

Eaton2022, 2023-02-14:
> ONCE dataset has no annos in test split. According to ONCE official website(https://once-for-auto-driving.github.io/), the test results can be obtained from codalab. But codalab stops their old servers several months ago. Till now I have no idea where to submit result

Thanks for your kind reply.

closed 2023-04-01 by github-actions[bot]