open-mmlab/OpenPCDet #1260: Can I directly use the pretrained model to predict my own data?

tub-sgg, 2023-02-08:
I want to direcltly use the pretrained model(i.e, the pretrained model is based on KITTI dataset), I also collected my own point cloud data( here only one frame maybe). Now i want check whether the model works or not.
if i directly use my own data, nothing could be detected.
So, what should I do to deal with my own data or some other configs
[image]

jihanyang, 2023-02-09:
Please refer to the source only results of UDA repo such as ST3D.

aminzadenoori, 2023-02-15:
Also this is my question too.

can you please explain on more details how can I use pretrained model to extract bounding boxes of cars on 3d point clouds?

closed 2023-04-02 by github-actions[bot]