open-mmlab/OpenPCDet #1261: pvrcnn++ code

suvasis, 2023-02-10:
hi,

Which files are relevant for the paper PVRCNN++ :
In the following location:
1. OpenPCDet/tree/master/pcdet/models/backbones_3d/
    besides the following files:
    a) spconv_backbone_focal.py
    a) pfe/voxel_set_abstraction.py

DezeZhao, 2023-02-12:
pfe/voxel_set_abstraction.py

QingXIA233, 2023-02-13:
@DezeZhao Hello, I tried to use PVRCNN++ on my own dataset. I have already changed the eval scrit according to my own dataset settings. Everything is all right. However, when I trained CenterPoint-Voxel and PVRCNN++ individually for 30 epochs on my own dataset, I got surprising results. The performance of PVRCNN++ is worse than that of CP-Voxel. My dataset has 11 classes instead of 3 in WOD. I wonder which parameters of PVRCNN++ I should tune to get better result than CP-Voxel?

closed 2023-03-31 by github-actions[bot]