open-mmlab/OpenPCDet #1263: 运行python train.py --cfg_file cfgs/kitti_models/pv_rcnn.yaml报错

6Superman6, 2023-02-13:
epochs:   0%| | 0/80 [00:21<?, ?it/s, loss=5.77, lr=0.001, d_time=0.00(0.16), f_time=1.05(1.04), b_time=1.72(2.55
Traceback (most recent call last):
  File "train.py", line 205, in <module>
    main()
  File "train.py", line 156, in main
    train_model(
  File "/home/dell/project/pcdet/GLENet-main/tools/train_utils/train_utils.py", line 133, in train_model
    accumulated_iter = train_one_epoch(
  File "/home/dell/project/pcdet/GLENet-main/tools/train_utils/train_utils.py", line 47, in train_one_epoch
    loss, tb_dict, disp_dict = model_func(model, batch)
  File "/home/dell/project/pcdet/GLENet-main/tools/../pcdet/models/__init__.py", line 42, in model_func
    ret_dict, tb_dict, disp_dict = model(batch_dict)
  File "/home/dell/anaconda3/envs/glenet/lib/python3.8/site-packages/torch/nn/modules/module.py", line 889, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/home/dell/project/pcdet/GLENet-main/tools/../pcdet/models/detectors/pv_rcnn.py", line 11, in forward
    batch_dict = cur_module(batch_dict)
  File "/home/dell/anaconda3/envs/glenet/lib/python3.8/site-packages/torch/nn/modules/module.py", line 889, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/home/dell/project/pcdet/GLENet-main/tools/../pcdet/models/roi_heads/pvrcnn_head.py", line 146, in forward
    targets_dict = self.assign_targets(batch_dict)
  File "/home/dell/project/pcdet/GLENet-main/tools/../pcdet/models/roi_heads/roi_head_template.py", line 134, in assign_targets
    targets_dict = self.proposal_target_layer.forward(batch_dict)
  File "/home/dell/project/pcdet/GLENet-main/tools/../pcdet/models/roi_heads/target_assigner/proposal_target_layer.py", line 32, in forward
    batch_rois, batch_gt_of_rois, batch_roi_ious, batch_roi_scores, batch_roi_labels, batch_gt_uncertaintys_of_rois = self.sample_rois_for_rcnn(
  File "/home/dell/project/pcdet/GLENet-main/tools/../pcdet/models/roi_heads/target_assigner/proposal_target_layer.py", line 116, in sample_rois_for_rcnn
    sampled_inds = self.subsample_rois(max_overlaps=max_overlaps)
  File "/home/dell/project/pcdet/GLENet-main/tools/../pcdet/models/roi_heads/target_assigner/proposal_target_layer.py", line 170, in subsample_rois
    raise NotImplementedError
NotImplementedError

6Superman6, 2023-02-14:
> We recommend using English or English & Chinese for issues so that we could have broader discussion.

How to deal?

closed 2023-04-01 by github-actions[bot]

zbaishancha, 2023-06-07:
@6Superman6 Hi, I'm having the same problem as you, the error is reported at`pcdet/models/roi_heads/target_assigner/proposal_target_layer.py`中
[image]
I'm guessing it might be the code here
[image]
The number of roi selected is the nan value, so this error is reported. Have you solved it?

Zixiu99, 2024-08-29:
@Camellia-hz 
May I ask if you've solved it? I had the same problem.