open-mmlab/OpenPCDet #1279: Centerpoint and pillarnet on KITTI has 0 accuracy

ahmedius2, 2023-02-25:
I have been training pillarnet and centerpoint (custom config I found [here](https://github.com/open-mmlab/OpenPCDet/issues/750)) using KITTI and they appear training well, the loss decreases down to 1.0 or below. However, the evaluation results are very bad. I also trained pointpillars with KITTI and it is fine. What might be the problem? In the past, I trained the centerpoint with KITTI with the custom config given above and it worked very well. Did some commits since then break something?

zhangkangkai, 2023-02-28:
I also encountered this problem

closed 2023-04-15 by github-actions[bot]

zhanghaozhe23, 2023-06-19:
> I have been training pillarnet and centerpoint (custom config I found [here](https://github.com/open-mmlab/OpenPCDet/issues/750)) using KITTI and they appear training well, the loss decreases down [...]

have you solved the problem?

zhanghaozhe23, 2023-06-19:
> I also encountered this problem

have you solved the problem?

lacie-life, 2024-11-30:
Have you solved the problem? @zhangkangkai