open-mmlab/OpenPCDet #1282: Reduce the number of samples

amineboujnah-dev, 2023-02-27:
Hello how can I modify the configuration of files in order to reduce the number of samples for both training and testing ?

amineboujnah-dev, 2023-02-27:
Thanks, I will try it.
Can you explain the difference between data training/testing examples and samples ?

amineboujnah-dev, 2023-02-27:
So the testing set of kitti is used for what exactly ?

ahmedius2, 2023-02-27:
It's not used for training or evaluation as it doesn't have annotations. However, it can be used for testing a model in general, without calculating the accuracy.

carlvinson70, 2023-03-03:
> Yes, it is possible. For KITTI, if you go to OpenPCDET/data/kitti/ImageSets/ folder, there are files with the list of samples to be processed. You can play with them and then regenerate the .pkl files, that should work. The same technique should also work for Waymo. For nuscenes, you need to modify [the splits.py file of the nuscenes-devkit](https://github.com/nutonomy/nuscenes-devkit/blob/master/python-sdk/nuscenes/utils/splits.py). Each scene corresponds to ~40 samples. After changing it, you need to regenerate the pkl files as explained in the "Getting Started" part of the readme.

Hi, I'm just wondering if it is possible to sample and load a customised amount of training data for training by setting Dataset or DataLoader, without regenerating a new pkl each time?

closed 2023-03-11 by amineboujnah-dev