open-mmlab/OpenPCDet #1283: AttributeError: 'EasyDict' object has no attribute 'BACKUP_DB_INFO'

zheng0819, 2023-02-28:
When I'm running the code in debug mode, an error has occurred :"File "/media/hddspace/hys/OpenPCDet/pcdet/datasets/augmentor/database_sampler.py", line 35, in __init__
    sampler_cfg.DB_INFO_PATH[0] = sampler_cfg.BACKUP_DB_INFO['DB_INFO_PATH']
AttributeError: 'EasyDict' object has no attribute 'BACKUP_DB_INFO'". This error occurred when running the function "build_dataloader" at line 111 in train.py. I tried to update EasyDict but it did not work. How can I fix this problem? Thank you.

zheng0819, 2023-03-01:
I have solved it, this issue can be closed.

Paraboy-zt, 2023-03-02:
请问怎么解决的 谢谢啦

Paraboy-zt, 2023-03-02:
Could you tell me how to solve the problem？Thanks

Paraboy-zt, 2023-03-03:
> I have solved it, this issue can be closed.
Could you tell me how to solve the problem？Thanks

zmsdcy, 2023-03-15:
> When I'm running the code in debug mode, an error has occurred :"File "/media/hddspace/hys/OpenPCDet/pcdet/datasets/augmentor/database_sampler.py", line 35, in **init** sampler_cfg.DB_INFO_PATH[0] = [...]

Could you tell me how to solve the problem？Thanks

zmsdcy, 2023-03-15:
> > I have solved it, this issue can be closed.
> > Could you tell me how to solve the problem？Thanks

Could you tell me how to solve the problem？Thanks

lmn1516, 2023-04-26:
I have the same problem. Could you tell me how to solve the problem？Thanks

> > > I have solved it, this issue can be closed.
> > > Could you tell me how to solve the problem？Thanks
> 
> Could you tell me how to solve the problem？Thanks

closed 2023-06-09 by github-actions[bot]

Petros626, 2024-10-12:
@zheng0819 don't act like this

zheng0819, 2024-10-16:
I am so sorry that I didn't check my email very often so that I missed these questions. This error occured because that your path of the dataset may be not correct(try to use the absolute path). It is not about the version of Easydict. I sincerely apologize for my delayed response.