open-mmlab/OpenPCDet #1284: Error in eval pillarnet: KeyError: 'final_box_dicts'

XiangZhou1, 2023-03-05:
when i eval pillarnet, there is a error

File "train.py", line 231, in <module>
    main()
  File "train.py", line 224, in main
    dist_test=dist_train
  File "/media/10T/zhouxiang/research/NewTA6_5/OpenPCDet/tools/test.py", line 125, in repeat_eval_ckpt
    result_dir=cur_result_dir
  File "/media/10T/zhouxiang/research/NewTA6_5/OpenPCDet/tools/eval_utils/eval_utils.py", line 65, in eval_one_epoch
    pred_dicts, ret_dict = model(batch_dict)
  File "/media/10T/CRH/env/part_a2/lib/python3.7/site-packages/torch/nn/modules/module.py", line 727, in _call_impl
    result = self.forward(*input, **kwargs)
  File "../pcdet/models/detectors/pillarnet.py", line 21, in forward
    pred_dicts, recall_dicts = self.post_processing(batch_dict)
  File "../pcdet/models/detectors/pillarnet.py", line 39, in post_processing
    final_pred_dict = batch_dict['final_box_dicts']
KeyError: 'final_box_dicts'

how can i fix it

zhangkangkai, 2023-03-08:
If you use pillarnet+AnchorHeadSingle, the post-processing function in the pillarnet.py file should be commented out
`from .detector3d_template import Detector3DTemplate

class PillarNet(Detector3DTemplate):
    def __init__(self, model_cfg, num_class, dataset):
        super().__init__(model_cfg=model_cfg, num_class=num_class, dataset=dataset)
        self.module_list = self.build_networks()

    def forward(self, batch_dict):
        for cur_module in self.module_list:
            batch_dict = cur_module(batch_dict)

        if self.training:
            loss, tb_dict, disp_dict = self.get_training_loss()

            ret_dict = {
                'loss': loss
            }
            return ret_dict, tb_dict, disp_dict
        else:
            pred_dicts, recall_dicts = self.post_processing(batch_dict)
            return pred_dicts, recall_dicts

    def get_training_loss(self):
        disp_dict = {}

        loss_rpn, tb_dict = self.dense_head.get_loss()
        tb_dict = {
            'loss_rpn': loss_rpn.item(),
            **tb_dict
        }

        loss = loss_rpn
        return loss, tb_dict, disp_dict
`

closed 2023-04-22 by github-actions[bot]

yuezhao1, 2023-05-25:
@zhangkangkai  Thank you very much. After commenting it out, I can evaluate the pillarnet model normally

luo3216, 2023-07-15:
the comment already exits, it doesn't work for me

luo3216, 2023-07-15:
@XiangZhou1 Thanks a lot. it works for me .