open-mmlab/OpenPCDet #1286: Help："pickle data was truncated" when continue training on nuScenes

littleship123, 2023-03-07:
I used "ctrl+C" to interrupt the existing training, and adjusted the ‘’-- workers‘’ parameter (some blogs said that this can speed up training, when the GPU utilization is low)

When I use the same script file to train from checkpoint again (OpenPCDet will automatically read the latest checkpoint file), the following error occurs:
`
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/dyz/anaconda3/envs/OpenPCDet/lib/python3.8/multiprocessing/spawn.py", line 116, in spawn_main
    exitcode = _main(fd, parent_sentinel)
  File "/home/dyz/anaconda3/envs/OpenPCDet/lib/python3.8/multiprocessing/spawn.py", line 126, in _main
    self = reduction.pickle.load(from_parent)
_pickle.UnpicklingError: pickle data was truncated
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/dyz/anaconda3/envs/OpenPCDet/lib/python3.8/multiprocessing/spawn.py", line 116, in spawn_main
    exitcode = _main(fd, parent_sentinel)
  File "/home/dyz/anaconda3/envs/OpenPCDet/lib/python3.8/multiprocessing/spawn.py", line 126, in _main
    self = reduction.pickle.load(from_parent)
_pickle.UnpicklingError: pickle data was truncated
Exception ignored in: <function DataBaseSampler.__del__ at 0x7f17c8ab9550>
Traceback (most recent call last):
  File "/home/dyz/OpenPCDet-master/tools/../pcdet/datasets/augmentor/database_sampler.py", line 73, in __del__
    if self.use_shared_memory:
AttributeError: 'DataBaseSampler' object has no attribute 'use_shared_memory'
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/dyz/anaconda3/envs/OpenPCDet/lib/python3.8/multiprocessing/spawn.py", line 116, in spawn_main
    exitcode = _main(fd, parent_sentinel)
  File "/home/dyz/anaconda3/envs/OpenPCDet/lib/python3.8/multiprocessing/spawn.py", line 126, in _main
    self = reduction.pickle.load(from_parent)
_pickle.UnpicklingError: pickle data was truncated
`
The screenshot
[image]

Even if the default ‘’-- workers‘’ parameter is applied again, the error still exists.
More Strangely, even if you execute a complete new training program or delete all checkpoint files instead of reading existing checkpoints, it will report the same error. I'm sure the server has enough space (in fact, the server was more crowded when I first ran the training script.)

I don't understand why the first training is normal, but now both the new one and the old one can't be trained. Will interruption of a 
 training damage the dataset files?

WoodwindHu, 2023-03-29:
same question

jetpackfirstme, 2023-04-05:
Same question. I can train with a single GPU, but when trying multi GPUs, the programs reports the same error.

jetpackfirstme, 2023-04-05:
> 

I just found the cause of this error. The cause is that RAM is not enough. I checked the free RAM while I run the experiment and found that when the error occurred, there is almost no free RAM left. So I tried to reduce the num_workers to 1, and the experiment can run normally, except that the speed is not very fast. Hope this useful for you.

jetpackfirstme, 2023-04-05:
> Same question. I can train with a single GPU, but when trying multi GPUs, the programs reports the same error.

I just found the cause of this error. The cause is that RAM is not enough. I checked the free RAM while I run the experiment and found that when the error occurred, there is almost no free RAM left. So I tried to reduce the num_workers to 1, and the experiment can run normally, except that the speed is not very fast. Hope this useful for you.

closed 2023-05-20 by github-actions[bot]