open-mmlab/OpenPCDet #1293: AttributeError: 'EasyDict' object has no attribute 'BACKUP_DB_INFO'

zmsdcy, 2023-03-16:
I came completely according to the tutorial and didn't modify the code. I don't know if this problem is related to the compatibility of the library version，thanks

zmsdcy, 2023-03-16:
I have solved this problem and can close it. It can be solved through the command line of
  conda install - c conda - forge easydict

CopleManba, 2023-04-16:
> I have solved this problem and can close it. It can be solved through the command line of conda install - c conda - forge easydict

I have tried your solution, but it did not solved. Did you modified other place?

CaoJiahao2, 2023-04-16:
I also meet this problem,and I wonder what cause this problem.I believe that this is not my fault.

CopleManba, 2023-04-18:
> I also meet this problem,and I wonder what cause this problem.I believe that this is not my fault.

Did you solve it? I really do not know how to solve it ! ! !

apecout, 2023-04-20:
I had the same issue. The *.pkl files names in pointpillar.yaml and mydataset.yaml didn't match files created in data preparation.

beyoursel, 2023-05-12:
[image]
maybe changing the code in the picture will  work

beyoursel, 2023-05-12:
[image]
maybe change the code in the picture will work(in my project, I use absolute path of Dataset)

Clichong, 2023-06-04:
I meet the same question, take care of the path, and you will solve it.

beyoursel, 2023-06-04:
您的邮件已收到，我会尽快阅读！

closed 2023-07-19 by github-actions[bot]

zhu2900000, 2024-12-06:
> This issue was closed because it has been inactive for 14 days since being marked as stale.

I think you may not make your own dataset on openpcdet,I have some issus, and I sovle it when I redo it.(I make creat dataset on other project)
[image]
[image]
And I am not sure the different of this dataset except name, I am recreate on Openpcdet