open-mmlab/OpenPCDet #1300: 只有点云数据进行训练出现如下错误：

mjc12326, 2023-03-25:
File "/home/mjc/OpenPCDet/tools/../pcdet/models/backbones_2d/map_to_bev/pointpillar_scatter.py", line 12, in __init__
    assert self.nz == 1
AssertionError

OrangeSodahub, 2023-03-25:
@mjc12326 You could refer to [this](https://github.com/OrangeSodahub/CRLFnet).
That `self.nz == 1` means you should make sure that your voxel size in the z-axis is equal to your height of the point cloud range.

yjwong1999, 2023-05-02:
> @mjc12326 You could refer to [this](https://github.com/OrangeSodahub/CRLFnet). That `self.nz == 1` means you should make sure that your voxel size in the z-axis is equal to your height of the point cloud range.

@mjc12326  I faced the same issue when configuring pointpillar for my custom dataset.
I managed to solve this issue as mentioned here.
Thanks a lot to @OrangeSodahub.

closed 2023-06-17 by github-actions[bot]

wmz555, 2025-03-27:
> [@mjc12326](https://github.com/mjc12326) I faced the same issue when configuring pointpillar for my custom dataset. I managed to solve this issue as mentioned here. Thanks a lot to [...]

Hello, I also encountered this problem while training a homebrew dataset with a pointpillar network, may I ask how you solved it in the end, I'm looking forward to your reply!