open-mmlab/OpenPCDet #1304: Setting MAX_NUMBER_OF_VOXELS

darrenjkt, 2023-03-27:
Hi, 

I have a question about max_number_of_voxels. In the waymo multiframe yaml and other configs, I noticed the training and testing max_number_of_voxels are different. Why is that?

https://github.com/open-mmlab/OpenPCDet/blob/31f6758af378c20e6200cd8e16307b1afa8c8a42/tools/cfgs/dataset_configs/waymo_dataset_multiframe.yaml#L83-L89

For the voxel setting above and a range of [-75.2, -75.2, -2, 75.2, 75.2, 4], we have the number of voxels as [150.4m/0.1, 150.4m/0.1, 6m/0.15]=[1504,1504,40]. This would give a total of 1504*1504*40=90,480,640.

I'm guessing we set a max number of voxels so we save the compute for 90 million voxels but why is the number different for training/testing? 

Does it affect performance if we want to fine-tune a pre-trained ckpt with the same voxel size but different max number of voxels to the pre-trained ckpt configs?

yukang2017, 2023-04-03:
Hi, based on my experience, for training, a smaller `MAX_NUMBER_OF_VOXELS` is to avoid out-of-memory. 
For testing, we use a larger one is to avoid that the performance might be compromised if there are scenes have more  voxel numbers than `MAX_NUMBER_OF_VOXELS` and the voxels might be removed.

darrenjkt, 2023-04-03:
I see, that makes sense! I was thinking it could be memory related. I did see some minor performance increase when I had more voxels for testing in some cases. Thanks

btryq, 2023-04-12:
@yukang2017 @darrenjkt Hello!! I am still curious about the MAX_NUMBER_OF_VOXELS here. Let's say you have 90,480,640 voxels in your detection range ([-75.2, -75.2, -2, 75.2, 75.2, 4]), but you only set 180,000 voxels for training and 400,000 for testing. 

So my question here is: how about the rest voxels? Have they been removed randomly when training and testing?