open-mmlab/OpenPCDet #1308: Train MPPNet With KITTI DATASET

GSNCodes, 2023-03-31:
Hello,

Is there a way to train new MPPNet architecture using KITTI Dataset?
Any guide or suggestions would help. Thanks.

yukang2017, 2023-04-03:
Hi, 

According to my understanding, MPPNet is for multi-frame detection, while the multi-frame data in KITTI is not temporally continuous. I think this might be hard to train.

Regards,
Yukang Chen

GSNCodes, 2023-04-03:
Thank you for responding @yukang2017 
I have a data in KITTI Format that's continuous. I was wondering whether it'd be possible to modify training in any way to use a different data format. 

Note: I don't have Ground Truth Velocity unlike the Waymo Dataset

closed 2023-05-18 by github-actions[bot]

Tottowich, 2023-07-16:
@GSNCodes Hi! I am also looking into using an MPPNet model on a custom KITTI format dataset. Have you made any progress on this?