open-mmlab/OpenPCDet #1311: How to get the KITTI [0-30)[30-50)[50-inf) type results using OpenPCDet?

YingjuanTang, 2023-04-04:
I notice there is some metric in KITTI by splitting the distance into [0-30) [30-50) [50 - ∞）
How to get the results in this form using OpenPCDet?

yukang2017, 2023-04-18:
Hi, OpenPCDet currently does not support this feature yet. But you can try this one

https://github.com/Abyssaledge/Extensible-Object-Detection-Evaluator

darrenjkt, 2023-05-01:
You could convert the prediction boxes to waymo format to use Waymo evaluation. Then you have to replace OBJECT TYPE with RANGE to get the metric breakdown by range.
https://github.com/open-mmlab/OpenPCDet/blob/ad9c25c03cf9e866bbee8c48fa4e2860cce68a9c/pcdet/datasets/waymo/waymo_eval.py#L94

Change to:
```
def build_config(self):
        config = metrics_pb2.Config()
        config_text = """
        breakdown_generator_ids: RANGE
        difficulties {
        levels:1
        levels:2
        }
        matcher_type: TYPE_HUNGARIAN
        iou_thresholds: 0.0
        iou_thresholds: 0.7
        iou_thresholds: 0.5
        iou_thresholds: 0.5
        iou_thresholds: 0.5
        box_type: TYPE_3D
        """

        for x in range(0, 100):
            config.score_cutoffs.append(x * 0.01)
        config.score_cutoffs.append(1.0)

        text_format.Merge(config_text, config)
        return config
```

closed 2023-06-16 by github-actions[bot]