open-mmlab/OpenPCDet #1317: Regarding the issue of training interruption and continued training, resulting in a decrease in accuracy

ylwhxht, 2023-04-16:
Thank you for your very diligent work. This project has greatly benefited me!
I found that during the model training process, if the checkpoint file that was previously saved is interrupted and reloaded to continue training, even if the model parameters and optimizer parameters are reloaded, the model accuracy will show a downward trend in the early stages of training or even for a relatively long epoch, and finally gradually increase.

Observing the code, I think it may be due to the fluctuation in learning rate caused by the use of cosine lr mechanism despite the recovery of optimizer parameters, and the initial high learning rate. I don't know if it's because of this or other issues.

yukang2017, 2023-04-18:
Hi, would you please provide us more detailed information about which config file you use?

In addition, would you please provide us the checkpoint? Thus, we can reproduce this issue for analysis.

ylwhxht, 2023-04-18:
> Hi, would you please provide us more detailed information about which config file you use?
> 
> In addition, would you please provide us the checkpoint? Thus, we can reproduce this issue for analysis.

Thank you very much for your kind reply.
I think I found the problem, I think it is the change in the calculation of the learning rate that I want to increase the number of Thank you very much for your kind reply.
I think I found the problem, I think it should be that when I continue training, I want to increase the number of epochs trained, which leads to changes in the calculation of the learning rate, which in turn reduces the accuracy of the initial training of the model.
If epoch does not change, the above problem will not occur.

> Hi, would you please provide us more detailed information about which config file you use?
> 
> In addition, would you please provide us the checkpoint? Thus, we can reproduce this issue for analysis.

Thank you very much for your kind reply.
I think I found the problem, I think it should be that when I continue training, I want to increase the number of epochs trained, which leads to changes in the calculation of the learning rate, which in turn reduces the accuracy of the initial training of the model.
If the expected epoch does not change, the above problems will not occur.

closed 2023-06-02 by github-actions[bot]

yang-jiaye, 2025-12-14:
> Thank you very much for your kind reply. I think I found the problem, I think it is the change in the calculation of the learning rate that I want to increase the number of Thank you very much for [...]

Thanks! I meet same problem, so how do you finally fix the problem? I mean how you increase epochs