open-mmlab/OpenPCDet #1318: RuntimeError: The expanded size of the tensor (8) must match the existing size (10) at non-singleton dimension 1.  Target sizes: [128, 8].  Tensor sizes: [128, 10]

luyao2001, 2023-04-17:
when I train PDV on waymo dataset, I meet this error. 
The yaml file is "cfgs/waymo_models/pdv.yaml". I have cost much time on this.

error details: 
Traceback (most recent call last):
  File "train.py", line 232, in <module>
    main()
  File "train.py", line 176, in main
    train_model(
  File "/home/rcvlab/lly/PDV/tools/train_utils/train_utils.py", line 86, in train_model
    accumulated_iter = train_one_epoch(
  File "/home/rcvlab/lly/PDV/tools/train_utils/train_utils.py", line 38, in train_one_epoch
    loss, tb_dict, disp_dict = model_func(model, batch)
  File "/home/rcvlab/lly/PDV/pcdet/models/__init__.py", line 42, in model_func
    ret_dict, tb_dict, disp_dict = model(batch_dict)
  File "/home/rcvlab/anaconda3/envs/lly_torch17/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1110, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/rcvlab/lly/PDV/pcdet/models/detectors/pdv.py", line 11, in forward
    batch_dict = cur_module(batch_dict)
  File "/home/rcvlab/anaconda3/envs/lly_torch17/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1110, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/rcvlab/lly/PDV/pcdet/models/roi_heads/pdv_head.py", line 239, in forward
    targets_dict = self.assign_targets(batch_dict)
  File "/home/rcvlab/lly/PDV/pcdet/models/roi_heads/roi_head_template.py", line 110, in assign_targets
    targets_dict = self.proposal_target_layer.forward(batch_dict)
  File "/home/rcvlab/lly/PDV/pcdet/models/roi_heads/target_assigner/proposal_target_layer.py", line 32, in forward
    batch_rois, batch_gt_of_rois, batch_roi_ious, batch_roi_scores, batch_roi_labels = self.sample_rois_for_rcnn(
  File "/home/rcvlab/lly/PDV/pcdet/models/roi_heads/target_assigner/proposal_target_layer.py", line 113, in sample_rois_for_rcnn
    batch_gt_of_rois[index] = cur_gt[gt_assignment[sampled_inds]]
RuntimeError: The expanded size of the tensor (8) must match the existing size (10) at non-singleton dimension 1.  Target sizes: [128, 8].  Tensor sizes: [128, 10]

closed 2023-06-01 by github-actions[bot]

KyleYueye, 2023-09-25:
Any update? Same question

jamesheatonrdm, 2024-06-04:
@KyleYueye solved [here](https://github.com/open-mmlab/OpenPCDet/issues/1084#issuecomment-1233119065)