open-mmlab/OpenPCDet #1324: Difference of "infer_time" and "sec_per_sample"

Rzlnmc, 2023-04-21:
Hi! Thank you for your excellent work.

I m just confused about "infer_time" and "sec_per_sample". 

When I use test.py to test one pth, "sec_per_example" is very low, like 0.0002 second, and "infer_time" is about  57.

When I use train.py, during "EVALUATION", "sec_per_example" is about 0.0113 second for each apoch.

So, what are the meanings of "infer_time" and "sec_per_sample" and which one can be used to calculate the FPS of the model?

Thanks.

closed 2023-06-09 by github-actions[bot]

jaan1729, 2023-10-03:
The "infer_time" measures the inference time of the deep learning model. It is the time taken for the step
`pred_dicts, ret_dict = model(batch_dict)'

On the other hand, "secs_per_example" is an average processing time for batch, accounts not only the model's inference time but also includes time for preprocessing, post-processing and evaluation steps.  It's always longer than the "infer_time"

bqcs, 2023-12-26:
> Hi! Thank you for your excellent work.
>
> I m just confused about "infer_time" and "sec_per_sample".
>
> When I use test.py to test one pth, "sec_per_example" is very low, like 0.0002 second, and [...]

hello,why cant't i find any infortmation related to "infer_time"  when i run the test.py file?

cat722, 2023-12-26:
@bqcs maybe you can use "--infer_time" when you run the test.py file

Rzlnmc, 2023-12-27:
> The "infer_time" measures the inference time of the deep learning model. It is the time taken for the step `pred_dicts, ret_dict = model(batch_dict)'
>
> On the other hand, "secs_per_example" is an [...]

 Thank you for your detailed answer. Sorry for late reply.