open-mmlab/OpenPCDet #1328: ModuleNotFoundError: No module named 'av2'

lmn1516, 2023-04-24:
Traceback (most recent call last):
  File "train.py", line 7, in <module>
    from test import repeat_eval_ckpt
  File "/home/l/OpenPCDet/tools/test.py", line 14, in <module>
    from eval_utils import eval_utils
  File "/home/l/OpenPCDet/tools/eval_utils/eval_utils.py", line 8, in <module>
    from pcdet.models import load_data_to_gpu
  File "../pcdet/models/__init__.py", line 6, in <module>
    from .detectors import build_detector
  File "../pcdet/models/detectors/__init__.py", line 12, in   ##<module>
    from .mppnet import MPPNet
  File "../pcdet/models/detectors/mppnet.py", line 9, in <module>
    from pcdet.datasets.augmentor import augmentor_utils, database_sampler
  File "../pcdet/datasets/__init__.py", line 15, in <module>
    from .argo2.argo2_dataset import Argo2Dataset
  File "../pcdet/datasets/argo2/argo2_dataset.py", line 7, in <module>
    from av2.utils.io import read_feather
ModuleNotFoundError: No module named 'av2'
我在开始训练的时候出现了这个错误，请问该怎么解决？

yukang2017, 2023-04-24:
This is because the new version supports Argoverse2 dataset, as this [document](https://github.com/open-mmlab/OpenPCDet/blob/master/docs/GETTING_STARTED.md#argoverse2-dataset). 

Please install this.
`pip install av2==0.2.0`

HezyC, 2023-05-13:
Hello! I encountered the same issue, and when testing demo. py, I was prompted that there was no 'av2' module available.

I tried commenting out the "from. argo2. argo2 dataset import Argo2Dataset" and "Argo2Dataset': Argo2Dataset " in "OpenPCDet/pcdet/datasets/__init__. py" to ensure normal operation.

Because I didn't download the Argo2Dataset dataset, I don't need this part, just comment it out. 
If you need it, you can try downloading the relevant dataset and installation package.

748811693aB, 2023-06-05:
> pip install av2==0.2.0

this helps，but I encounter a new problem:   File "/home/whut-4/Desktop/HXB/06-05/OpenPCDet/pcdet/utils/spconv_utils.py", line 4, in <module>
    if float(spconv.__version__[2:]) >= 2.2:
AttributeError: module 'spconv' has no attribute '__version__'

748811693aB, 2023-06-05:
> Hello! I encountered the same issue, and when testing demo. py, I was prompted that there was no 'av2' module available.
>
> I tried commenting out the "from. argo2. argo2 dataset import Argo2Dataset" [...]

this helps，but I encounter a new problem:   File "/home/whut-4/Desktop/HXB/06-05/OpenPCDet/pcdet/utils/spconv_utils.py", line 4, in <module>
    if float(spconv.__version__[2:]) >= 2.2:
AttributeError: module 'spconv' has no attribute '__version__'

HezyC, 2023-06-05:
Maybe you can check if the spconv installation is successful. I installed the spconv2. version, this problem did not occur

HezyC, 2023-06-05:
> > Hello! I encountered the same issue, and when testing demo. py, I was prompted that there was no 'av2' module available.
> > I tried commenting out the "from. argo2. argo2 dataset import Argo2Dataset" and "Argo2Dataset': Argo2Dataset " in "OpenPCDet/pcdet/datasets/**init**. py" to ensure normal operation.
> > Because I didn't download the Argo2Dataset dataset, I don't need this part, just comment it out. If you need it, you can try downloading the relevant dataset and installation package.
> 
> this helps，but I encounter a new problem: File "/home/whut-4/Desktop/HXB/06-05/OpenPCDet/pcdet/utils/spconv_utils.py", line 4, in if float(spconv.**version**[2:]) >= 2.2: AttributeError: module 'spconv' has no attribute '**version**'

Maybe you can check if the spconv installation was successful. I installed the spconv2. version, this problem did not occur.

748811693aB, 2023-06-05:
> Maybe you can check if the spconv installation was successful. I installed the spconv2. version, this problem did not occur.

thanks alot. I solved this by 
```
pip install spconv==2.*
```

CopleManba, 2023-06-11:
Maybe if you do not use  Argo2Dataset dataset, you can delete "from .argo2.argo2_dataset import Argo2Dataset".

HXB-1997, 2023-06-12:
> Maybe if you do not use Argo2Dataset dataset, you can delete "from .argo2.argo2_dataset import Argo2Dataset".

yep thanks

closed 2023-07-28 by github-actions[bot]