open-mmlab/OpenPCDet #1334: AttributeError: 'EasyDict' object has no attribute 'BACKUP_DB_INFO'

lmn1516, 2023-04-26:
Traceback (most recent call last):
  File "train.py", line 229, in <module>
    main()
  File "train.py", line 117, in main
    train_set, train_loader, train_sampler = build_dataloader(
  File "/home/l/OpenPCDet/tools/../pcdet/datasets/__init__.py", line 57, in build_dataloader
    dataset = __all__[dataset_cfg.DATASET](
  File "/home/l/OpenPCDet/tools/../pcdet/datasets/kitti/kitti_dataset.py", line 23, in __init__
    super().__init__(
  File "/home/l/OpenPCDet/tools/../pcdet/datasets/dataset.py", line 30, in __init__
    self.data_augmentor = DataAugmentor(
  File "/home/l/OpenPCDet/tools/../pcdet/datasets/augmentor/data_augmentor.py", line 23, in __init__
    cur_augmentor = getattr(self, cur_cfg.NAME)(config=cur_cfg)
  File "/home/l/OpenPCDet/tools/../pcdet/datasets/augmentor/data_augmentor.py", line 27, in gt_sampling
    db_sampler = database_sampler.DataBaseSampler(
  File "/home/l/OpenPCDet/tools/../pcdet/datasets/augmentor/database_sampler.py", line 35, in __init__
    sampler_cfg.DB_INFO_PATH[0] = sampler_cfg.BACKUP_DB_INFO['DB_INFO_PATH']
AttributeError: 'EasyDict' object has no attribute 'BACKUP_DB_INFO'

I met this problem when you start training, could you tell me how to solve?

yjwong1999, 2023-05-02:
Hi there, 
I faced this issue before, most likely is because you did not change the db_info in the config yaml file.

In my case, I'm using the pointpillar.yaml + custom dataset
Go to the yaml file, and change the `DB_INFO_PATH` to custom_dbinfos_train.pkl

[image]

You just have to modify the `DB_INFO_PATH` to wherever you saved the pkl files

CopleManba, 2023-06-14:
Just check your data path! I suggest you to create soft link to this project!

closed 2023-07-30 by github-actions[bot]

luo3216, 2023-08-02:
nothing to do with pakage. pls check your data path, wther your data exits or not.

Zixiu99, 2024-02-24:
I am getting exactly the same error reported, I can confirm that I have generated a valid bin file in gt_database and the DB_INFO_PATH is custom_dbinfos_train.pkl. is there anyone else who is facing the same problem as me?(And also commented out (#filter_by_difficulty: [-1])

Petros626, 2024-10-12:
@Zixiu99 for now, yes I'm facing the same using custom_dataset.py, but I'm using KITTI dataset with it.