open-mmlab/OpenPCDet #1335: Custom Dataset

yjwong1999, 2023-04-29:
Hi OpenPCDet teams, 

First and foremost, this is a fantastic repo! Absolutely appreciate the team's effort!

I'm implementing my custom dataset with pure point cloud. I have followed the guides on custom dataset. 
However, im an absolute beginner in 3D object detection. Im wondering what does the following terms mean:

1. What is the six values in `POINT_CLOUD_RANGE`
2. What is `NUM_POINT_FEATURES`in gt_sampling data augmentation

@jihanyang can you assist me on this? I saw you helping some issue regarding custom dataset. Many thanks in advance

jihanyang, 2023-04-30:
1. [x_min, y_min, z_min, x_max, y_max, z_max] in meters
2. the number of features of your point cloud, such as [x, y, z, intensity] in KITTI should be 4

yjwong1999, 2023-04-30:
Alright, thanks a lot! 
I have modified accordingly, and the code is running!
Many thanks

closed 2023-04-30 by yjwong1999

dikubab, 2023-05-25:
@yjwong1999 can you share the code you used to modify acordingly?