open-mmlab/OpenPCDet #1341: Encountered a parameter mismatch issue while running the pre training model

wangyakunn, 2023-05-16:
Hello, I've been able to follow the demo.py script to visualize the detections with KITTI .bin files and my own Lidar data using Pointpillar ( with the same config and pretrained model as in the DEMO.md).
Now I want to test the CenterPoint model of your model zoo trained in NuScenes with my own .npy files, but I get the following error:

`python3 demo.py --cfg_file cfgs/nuscenes_models/cbgs_dyn_pp_centerpoint.yaml --ckpt cbgs_pp_centerpoint_nds6070.pth --data_path xxxxxx.bin`

`2023-05-16 07:58:06,347   INFO  -----------------Quick Demo of OpenPCDet-------------------------
2023-05-16 07:58:06,347   INFO  Total number of samples:        1
2023-05-16 07:58:09,752   INFO  ==> Loading parameters from checkpoint ../data/kitti/cbgs_pp_centerpoint_nds6070.pth to CPU
2023-05-16 07:58:09,800   INFO  ==> Done (loaded 458/458)
Traceback (most recent call last):
  File "demo.py", line 112, in <module>
    main()
  File "demo.py", line 94, in main
    for idx, data_dict in enumerate(demo_dataset):
  File "demo.py", line 59, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)
  File "/home/kin/workspace/OpenPCDet/pcdet/datasets/dataset.py", line 180, in prepare_data
    data_dict = self.point_feature_encoder.forward(data_dict)
  File "/home/kin/workspace/OpenPCDet/pcdet/datasets/processor/point_feature_encoder.py", line 29, in forward
    data_dict['points'], use_lead_xyz = getattr(self, self.point_encoding_config.encoding_type)(
  File "/home/kin/workspace/OpenPCDet/pcdet/datasets/processor/point_feature_encoder.py", line 48, in absolute_coordinates_encoding
    assert points.shape[-1] == len(self.src_feature_list)
AssertionError`

I printed “points. shape [-1]” with a value of 4,I guess it's the x, y, z, intensity of the point cloud.While len(self.src_feature_list) is 5. 

I found this parameter in nuscenes_dataset.yaml: used_ feature_ List: ['x ',' y ',' z ',' intensity ',' timestamp '], “timestamp” is the additional parameterthis may be the reason for AssertionError. How can I modify the code to run through the Nuscenes dataset model?

jihanyang, 2023-05-17:
Can you test the ckpt normally?

wangyakunn, 2023-05-17:
> Can you test the ckpt normally?

I didn't perform the training, I just downloaded the trained model（cbgs_pp_centerpoint_nds6070.pth） and tried the effect of the model.

I don't understand why kitti works under the same operation, but nuscenes will report an error

wangyakunn, 2023-05-18:
I have reviewed question #195  and I have roughly understood some of it. The command line I used is: `python3 demo.py --cfg_file cfgs/nuscenes_models/cbgs_dyn_pp_centerpoint.yaml --ckpt cbgs_pp_centerpoint_nds6070.pth --data_path xxxxxx.bin`, data only uses one bin from the Nuscenes dataset. In question #195 , it seems that to run demo, the data format must be modified. What kind of data format is feasible? thank you very much！

wangyakunn, 2023-05-20:
> Can you test the ckpt normally?你能正常测试ckpt吗？

Hello, I have successfully conducted training and testing. How can I visualize the effect of a point cloud frame next

closed 2023-07-06 by github-actions[bot]

HAMA-DL-dev, 2023-08-24:
> I have reviewed question #195 and I have roughly understood some of it. The command line I used is: `python3 demo.py --cfg_file cfgs/nuscenes_models/cbgs_dyn_pp_centerpoint.yaml --ckpt [...]

I solved this problem by editing [nuscenes_dataset.yaml](https://github.com/open-mmlab/OpenPCDet/blob/bce886d6e36e3deaec2ce6f8b54b133fcadf62d6/tools/cfgs/dataset_configs/nuscenes_dataset.yaml#L57) like below
```yaml
POINT_FEATURE_ENCODING: {
    encoding_type: absolute_coordinates_encoding,
    #used_feature_list: ['x', 'y', 'z', 'intensity', 'timestamp'],
    #src_feature_list: ['x', 'y', 'z', 'intensity', 'timestamp'],
    used_feature_list: ['x', 'y', 'z', 'intensity'],
    src_feature_list: ['x', 'y', 'z', 'intensity'],
}
```
I also checked difference between point feautre and `self.src_feature_list` which is defined by `nuscenes_dataset.yaml` using code line ; `print(points.shape, len(self.src_feature_list))`

### run command 
```bash
python demo.py --cfg_file cfgs/nuscenes_models/cbgs_pp_multihead.yaml --ckpt ../checkpoints/nuScenes/pp_multihead_nds5823_updated.pth --data_path ../data/nuscenes/v1.0-mini/sweeps/LIDAR_TOP/n008-2018-08-01-15-16-36-0400__LIDAR_TOP__1533151610696833.pcd.bin
```

### result 

[image]

The result looks bad, but It is natural to expect poor results due to the domain difference(channel in LiDAR) between the custom dataset and the training dataset.

Though modified code, if there is the error still, mention me

yaobaishen, 2024-06-12:
@HAMA-DL-dev thanks for sharing your solution, but I think we shall not modify the nuscences_dataset.yaml, instead, I modify the demo.py as below:
```
    def __getitem__(self, index):
        if self.ext == '.bin':
            if self.is_nuscense_data:
                points = np.fromfile(self.sample_file_list[index],
                                     dtype=np.float32).reshape(-1, 5)
            else:
                points = np.fromfile(self.sample_file_list[index],
                                     dtype=np.float32).reshape(-1, 4)
```
Because the lidar data dimension of nuscenses data is 5.
And then we can run the demo.py, for example (I have added a --nuscense_data flag):
```
python demo.py --cfg_file cfgs/nuscenes_models/cbgs_dyn_pp_centerpoint.yaml --ckpt ../model_zoo/nuscenes_models/cbgs_pp_centerpoint_nds6070.pth --data_path dataset/nuscenes_mini/samples/LIDAR_TOP/n015-2018-11-21-19-38-26+0800__LIDAR_TOP__1542801004447480.pcd.bin --nuscense_data
```
but I still cannot get right inference result, it looks weird:
[image]
Please note that I have set inference score threshold to 0.3, there are still many unreasonable detection boxes.
I have create a new [issue_1627](https://github.com/open-mmlab/OpenPCDet/issues/1627) to track this problem, thanks.

yaobaishen, 2024-06-13:
> > I have reviewed question #195 and I have roughly understood some of it. The command line I used is: `python3 demo.py --cfg_file cfgs/nuscenes_models/cbgs_dyn_pp_centerpoint.yaml --ckpt cbgs_pp_centerpoint_nds6070.pth --data_path xxxxxx.bin`, data only uses one bin from the Nuscenes dataset. In question #195 , it seems that to run demo, the data format must be modified. What kind of data format is feasible? thank you very much！
> 
> I solved this problem by editing [nuscenes_dataset.yaml](https://github.com/open-mmlab/OpenPCDet/blob/bce886d6e36e3deaec2ce6f8b54b133fcadf62d6/tools/cfgs/dataset_configs/nuscenes_dataset.yaml#L57) like below
> 
> ```yaml
> POINT_FEATURE_ENCODING: {
>     encoding_type: absolute_coordinates_encoding,
>     #used_feature_list: ['x', 'y', 'z', 'intensity', 'timestamp'],
>     #src_feature_list: ['x', 'y', 'z', 'intensity', 'timestamp'],
>     used_feature_list: ['x', 'y', 'z', 'intensity'],
>     src_feature_list: ['x', 'y', 'z', 'intensity'],
> }
> ```
> 
> I also checked difference between point feautre and `self.src_feature_list` which is defined by `nuscenes_dataset.yaml` using code line ; `print(points.shape, len(self.src_feature_list))`
> 
> ### run command
> ```shell
> python demo.py --cfg_file cfgs/nuscenes_models/cbgs_second_multihead.yaml --ckpt ../checkpoints/nuScenes/cbgs_second_multihead_nds6229_updated.pth --data_path /home/hayeon/Tools_RosBag2KITTI/pcd2bin/bin/1638923805.193849087.bin
> ```
> 
> ### result
> ![image](https://private-user-images.githubusercontent.com/59410853/262951606-c0b1c81c-04ca-4be2-9394-012b52e41fc6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTgyNDE5OTEsIm5iZiI6MTcxODI0MTY5MSwicGF0aCI6Ii81OTQxMDg1My8yNjI5NTE2MDYtYzBiMWM4MWMtMDRjYS00YmUyLTkzOTQtMDEyYjUyZTQxZmM2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjEzVDAxMjEzMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWNhMTMwNjcxYTE2Y2QxZTFmNjRiZDkwZWE4M2ExNjFiZWI0ZWM5ZjUxYWQ2NmE0MTIzNzUyM2VkMDdhMGNlNjImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.xUBNSw1CN3TPq6R2bWP3bkCHNOaNOKDpKsBi94YJnPs)
> 
> The result looks bad, but It is natural to expect poor results due to the domain difference(channel in LiDAR) between the custom dataset and the training dataset.
> 
> If there is still an error despite the modified code, please notify me.

@HAMA-DL-dev Are you able to get correct inference result with the nuscenes_models? I have test with nuscenes_mini data with no luck.