open-mmlab/OpenPCDet #1344: OpenPCDet installation CUDA10.1//pytorch version//Compatibility

HackerHakcer, 2023-05-16:
Hello friends,
I am trying to install OpenPCDet on Ubuntu 20.04.4 LTS.
And i am running into error when i try to train PV-RCNN on a custom dataset using this command " python3 /home/nawfal/OpenPCDet/tools/train.py --cfg_file /home/nawfal/OpenPCDet/tools/cfgs/custom_models/pv_rcnn.yaml --batch_size 4 --epochs 50"
and i run into this error beIlow.
I think it is a problem of compatibility i want to know the version of pytorch that i need to install for cuda10.1.
If it is not the case of error please correct it to me.
Traceback (most recent call last):
  File "/home/nawfal/OpenPCDet/tools/train.py", line 7, in <module>
    from test import repeat_eval_ckpt
  File "/home/nawfal/OpenPCDet/tools/test.py", line 14, in <module>
    from eval_utils import eval_utils
  File "/home/nawfal/OpenPCDet/tools/eval_utils/eval_utils.py", line 8, in <module>
    from pcdet.models import load_data_to_gpu
  File "/home/nawfal/OpenPCDet/pcdet/models/__init__.py", line 6, in <module>
    from .detectors import build_detector
  File "/home/nawfal/OpenPCDet/pcdet/models/detectors/__init__.py", line 1, in <module>
    from .detector3d_template import Detector3DTemplate
  File "/home/nawfal/OpenPCDet/pcdet/models/detectors/detector3d_template.py", line 6, in <module>
    from ...ops.iou3d_nms import iou3d_nms_utils
  File "/home/nawfal/OpenPCDet/pcdet/ops/iou3d_nms/iou3d_nms_utils.py", line 9, in <module>
    from . import iou3d_nms_cuda
ImportError: /home/nawfal/OpenPCDet/pcdet/ops/iou3d_nms/iou3d_nms_cuda.cpython-38-x86_64-linux-gnu.so: undefined symbol: _ZNK2at6Tensor8data_ptrIfEEPT_v

jihanyang, 2023-05-17:
Try to re-install pcdet might help.

closed 2023-07-01 by github-actions[bot]

OuYaozhong, 2024-09-02:
> Try to re-install pcdet might help.

The same.

CUDA 11.8
Python 3.11
PyTorch 2.4
Ubuntu 22.04

Rebuild is useless. 
Rebuild and then the problem occur again.

Also followed [#216](https://github.com/open-mmlab/OpenPCDet/issues/216#issuecomment-669663092), useless too.

And it is a strange results that, if I try to rebuild, it will tell me that no 'CUDA_HOME' is specified.
But I have set it, because I have built it just now.

```
Traceback (most recent call last):
  File "/home/{$USER}//tmp/OpenPCDet/setup.py", line 59, in <module>
    make_cuda_ext(
  File "/home/{$USER}//tmp/OpenPCDet/setup.py", line 18, in make_cuda_ext
    cuda_ext = CUDAExtension(
               ^^^^^^^^^^^^^^
  File "/home/{$USER}/.conda/envs/{$ENV_NAME}/lib/python3.11/site-packages/torch/utils/cpp_extension.py", line 1076, in CUDAExtension
    library_dirs += library_paths(cuda=True)
                    ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/yiuzon/.conda/envs/{$ENV_NAME}/lib/python3.11/site-packages/torch/utils/cpp_extension.py", line 1207, in library_paths
    if (not os.path.exists(_join_cuda_home(lib_dir)) and
                           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/{$USER}/.conda/envs/{$ENV_NAME}/lib/python3.11/site-packages/torch/utils/cpp_extension.py", line 2416, in _join_cuda_home
    raise OSError('CUDA_HOME environment variable is not set. '
OSError: CUDA_HOME environment variable is not set. Please set it to your CUDA install root.
```

OuYaozhong, 2024-09-02:
> > Try to re-install pcdet might help.
> 
> The same.
> 
> CUDA 11.8 Python 3.11 PyTorch 2.4 Ubuntu 22.04
> 
> ```
> -@hostname:~/tmp/OpenPCDet$ nvcc -V
> nvcc: NVIDIA (R) Cuda compiler driver
> Copyright (c) 2005-2022 NVIDIA Corporation
> Built on Wed_Sep_21_10:33:58_PDT_2022
> Cuda compilation tools, release 11.8, V11.8.89
> Build cuda_11.8.r11.8/compiler.31833905_0
> -@ hostname:~/tmp/OpenPCDet $ conda list | grep spconv
> spconv-cu118              2.3.6                    pypi_0    pypi
> -@hostname:~/tmp/OpenPCDet$ gcc --version
> gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
> Copyright (C) 2021 Free Software Foundation, Inc.
> This is free software; see the source for copying conditions.  There is NO
> warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
> ```
> 
> Rebuild is useless. Rebuild and then the problem occur again.
> 
> Also followed [#216](https://github.com/open-mmlab/OpenPCDet/issues/216#issuecomment-669663092), useless too.
> 
> Error is shown below:
> 
> ```
> Traceback (most recent call last):
>   File "/home/{$USER}/tmp/OpenPCDet/tools/train.py", line 7, in <module>
>     from test import repeat_eval_ckpt
>   File "/home/{$USER}/tmp/OpenPCDet/tools/test.py", line 14, in <module>
>     from eval_utils import eval_utils
>   File "/home/{$USER}/tmp/OpenPCDet/tools/eval_utils/eval_utils.py", line 8, in <module>
>     from pcdet.models import load_data_to_gpu
>   File "/home/{$USER}/tmp/OpenPCDet/tools/../pcdet/models/__init__.py", line 6, in <module>
>     from .detectors import build_detector
>   File "/home/{$USER}/tmp/OpenPCDet/tools/../pcdet/models/detectors/__init__.py", line 1, in <module>
>     from .detector3d_template import Detector3DTemplate
>   File "/home/{$USER}/tmp/OpenPCDet/tools/../pcdet/models/detectors/detector3d_template.py", line 6, in <module>
>     from ...ops.iou3d_nms import iou3d_nms_utils
>   File "/home/{$USER}/tmp/OpenPCDet/tools/../pcdet/ops/iou3d_nms/iou3d_nms_utils.py", line 9, in <module>
>     from . import iou3d_nms_cuda
> ImportError: /home/{$USER}/tmp/OpenPCDet/tools/../pcdet/ops/iou3d_nms/iou3d_nms_cuda.cpython-311-x86_64-linux-gnu.so: undefined symbol: _ZN3c106detail14torchCheckFailEPKcS2_jRKSs
> ERROR conda.cli.main_run:execute(125): `conda run python /home/{$USER}/tmp/OpenPCDet/tools/train.py --cfg_file tools/cfgs/kitti_models/my_model.yaml --batch_size 16 --epochs 32` failed. (See above for error)
> ```
> 
> And it is a strange results that, if I try to rebuild, it will tell me that no 'CUDA_HOME' is specified. But I have set it, because I have built it just now.
> 
> ```
> Traceback (most recent call last):
>   File "/home/{$USER}//tmp/OpenPCDet/setup.py", line 59, in <module>
>     make_cuda_ext(
>   File "/home/{$USER}//tmp/OpenPCDet/setup.py", line 18, in make_cuda_ext
>     cuda_ext = CUDAExtension(
>                ^^^^^^^^^^^^^^
>   File "/home/{$USER}/.conda/envs/{$ENV_NAME}/lib/python3.11/site-packages/torch/utils/cpp_extension.py", line 1076, in CUDAExtension
>     library_dirs += library_paths(cuda=True)
>                     ^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/home/yiuzon/.conda/envs/{$ENV_NAME}/lib/python3.11/site-packages/torch/utils/cpp_extension.py", line 1207, in library_paths
>     if (not os.path.exists(_join_cuda_home(lib_dir)) and
>                            ^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/home/{$USER}/.conda/envs/{$ENV_NAME}/lib/python3.11/site-packages/torch/utils/cpp_extension.py", line 2416, in _join_cuda_home
>     raise OSError('CUDA_HOME environment variable is not set. '
> OSError: CUDA_HOME environment variable is not set. Please set it to your CUDA install root.
> ```

I have fixed this problem by using both `gcc=11.4` and `g++=11.4`

This problem maybe caused by the `gcc>11` or the version of `gcc` and `g++` is mismatch.

I finally fixed by using `g++=11.4`, because before that, my `gcc=11.4` but `g++=12`