open-mmlab/OpenPCDet #1349: How do I commit to the existing openpcdet framework?

zbaishancha, 2023-05-17:
I finished updating the openpcdet framework to support the [EPNet](https://arxiv.org/abs/2007.08856) model. 
I would like to share this work in openpcdet.
How do I go about it?

chenshi3, 2023-05-17:
Maybe you can submit a PR?

zbaishancha, 2023-05-17:
> 
Thanks. I'm going to try to submit it.

closed 2023-07-01 by github-actions[bot]

continue-up, 2025-03-12:
> I finished updating the openpcdet framework to support the [EPNet](https://arxiv.org/abs/2007.08856) model. I would like to share this work in openpcdet. How do I go about it?

Hello friend, I've also been looking into EPNet's multi-class detection recently. I'm really impressed by your deployment of EPNet to the OpenPCDet framework. This work is amazing. Could you share the specific solution? Or could you share your deployment code? If possible, my email is 3399730062@qq.com.