open-mmlab/OpenPCDet #1351: How to calculate bbox metric in custom data?

ies0411, 2023-05-19:
Hello. I am wondering about the way of bbox metric calculation in case of custom dataset.

There is no calibration and camera information in custom dataset format.

Then, I can not understand how to evaluate of bbox AP.

As I know, bbox AP is eval metric on the camera plane.

Is there anyone who can explain this??

jihanyang, 2023-05-22:
You need to define your own calibration file and implement the projection similar to kitti. You can just refer to how kitti dataset implement it.

ies0411, 2023-05-22:
Thank you for your reply.

But my question is something different from yours.

There is no calibration information in the custom data format (below)
[image]

So, I followed that format and converted my custom data and training successfully.
The result is below picture.
[image]

What I'm curious about is how did calculate bbox AP without calibration information??

dikubab, 2023-05-25:
@ies0411 can you share the code you used to generate .npy files?

ies0411, 2023-05-30:
@dikubab  I just followed the tutorial

jihanyang, 2023-05-31:
@ies0411 https://github.com/open-mmlab/OpenPCDet/blob/02ac3e17a9aa438bdab9b566d4334bff6263374a/pcdet/datasets/kitti/kitti_utils.py#L24
The box is given to a certain magic number.

ies0411, 2023-06-07:
@jihanyang 
Thanks. I totally understood.

ies0411, 2023-06-07:
I understand and I will close.

closed 2023-06-07 by ies0411

yinheyanxian, 2024-04-23:
Hello!
How do you evaluate without images?
I tried to evaluate using only the point cloud and found that I needed to comment out the code for the difficulty division of the KITTI dataset. How did you do that?
Thanks in advance!