open-mmlab/OpenPCDet #1354: Warning on RTX4090

zhangtingyu11, 2023-05-22:
I train the model on RTX4090,and it always ouput warning in the log.
```
[WARNING]your gpu arch (8, 9) isn't compiled in prebuilt, may cause invalid device function. available: {(6, 1), (7, 0), (8, 0), (8, 6), (6, 0), (7, 5), (5, 2)}
```
Does pcdet support arch8.9, or if it doesn't support, how can i fix this warning?

jihanyang, 2023-05-22:
Actually, we haven't tried on RTX4090. Which command that you run when raise this warning?

zhangtingyu11, 2023-05-22:
> Actually, we haven't tried on RTX4090. Which command that you run when raise this warning?

I found it's not the reason of OpenPCDet, it's the warning of spconv. I update the cuda version to 11.8 and install spconv-cu118, the warning disappear.Thank you for your reply.

closed 2023-05-22 by zhangtingyu11

RhythmOfTheRain-Byte, 2023-08-29:
> 
Excuse me
I'm using  RTX4090 too and may i ask what version of openpcdet do you have installed?
I am using openpcdet-v0.3
torch 1.12.1+cu113
spconv-cu113  2.3.3
When I run python setup.py develop, I get some error information，I guess which was caused by the version.

RhythmOfTheRain-Byte, 2023-08-29:
> I train the model on RTX4090,and it always ouput warning in the log.
>
> ```
> [WARNING]your gpu arch (8, 9) isn't compiled in prebuilt, may cause invalid device function. available: {(6, 1), (7, 0), (8, [...]

Excuse me
I'm using RTX4090 too and may i ask what version of openpcdet do you have installed?
I am using openpcdet-v0.3
torch 1.12.1+cu113
spconv-cu113 2.3.3
When I run python setup.py develop, I get some error information，I guess which was caused by the version.
Thank you!

zhangtingyu11, 2023-08-31:
> Excuse me I'm using RTX4090 too and may i ask what version of openpcdet do you have installed? I am using openpcdet-v0.3 torch 1.12.1+cu113 spconv-cu113 2.3.3 When I run python setup.py develop, I [...]

I  am using openpcdet-v0.6.0
torch 2.0.1+cu118
spcouv-cu118 2.3.6

sgrrlll, 2023-09-05:
> I am using openpcdet-v0.6.0 torch 2.0.1+cu118 spcouv-cu118 2.3.6

> I am using openpcdet-v0.6.0 torch 2.0.1+cu118 spcouv-cu118 2.3.6

这里都能遇见小黑子？？nb查了大半天，最后把spconv库换了就好了